riscv/riscv-cfi #3: `sspush` / `sspop`: Encoding ambiguity

a4lg, 2022-11-21:
Two instructions: `sspush` and `sspop` shares the same encoding except:
*   `sspush`: `rs1`==source, `rd`==0
*   `sspop`: `rs1`==0, `rd`==destination

If both `rs1` and `rd` are zero, `sspush x0` and `sspop x0` are indistinguishable.

Quoting "4.1. Push to and Pop from the shadow stack":
> the CFI extension provides the `x1` and `x5` variants of the shadow stack load and store.

It (this sentence from a note) suggests that `sspush` and `sspop` only allows either `x1` or `x5` but this is not clear from the specification.

How do we resolve this?  Here are some ideas to resolve:

1.  Only allow `x1` and `x5` as `sspush` and `sspop` instruction operand
2.  Disallow `x0` on `sspush` and `sspop` instructions (but allow other registers)
3.  Use different encodings (e.g. use different `funct3`) between `sspush` and `sspop` instructions
4.  Define the special behavior (explicitly) if an operand of `sspush` or `sspop` is `x0` (I don't like this option though)

ved-rivos, 2022-11-21:
using `rd=x0` is disallowed for `sspop` and using `rs1=x0` is disallowed for `sspush`. Other registers are allowed.

a4lg, 2022-11-22:
Thanks! (going forward with option 2)

ved-rivos, 2022-11-22:
Thanks. Document updated.

closed 2022-11-22 by ved-rivos