riscv/riscv-cfi #6: `sschkra`: strange encoding or operand ordering

a4lg, 2022-11-21:
Despite that `sschkra` is encoded as `rs1==5` and `rs2==1`, the only example of this instruction is used as: `sschkra x1, x5`.
If two operands are encoded as `rs1` and `rs2`, it usually means `rs1==1` and `rs2==5`.  This is rather strange.

Changing either encoding or operands of this instruction may be required.
Note that not all options are mutually exclusive (e.g. 1 and 3 are not mutually exclusive).

1.  Change encoding of `sschkra` so that `rs1==1` and `rs2==5`.
2.  Accept only `sschkra x5, x1` and change the example in the specification.
3.  Accept both `sschkra x1, x5` and `sschkra x5, x1`, assign them the same opcode.
4.  Remove operands from assembler mnemonics of `sschkra` and make `x1` and `x5` implicit.

ved-rivos, 2022-11-21:
So `lpsul`/`lpcul` use funct7 of `1000011` while `sschkra` uses `1000101` so there is no crash between them. 
The example should be fixed to match the intended encoding.

a4lg, 2022-11-22:
Thanks! (going forward with option 2)

> So `lpsul`/`lpcul` use funct7 of `1000011` while `sschkra` uses `1000101` so there is no crash between them.

It conflicts with the specification document.  It seems the document is wrong.  I submitted the fix (#7).

ved-rivos, 2022-11-22:
Thanks for the PR. Doc updated with example fixed.

closed 2022-11-22 by ved-rivos