riscv/riscv-cfi #30: Errors on atomic swap example

henry-hsieh, 2023-02-03:
The backward-edge CFI provides following example for atomic swap.
```
# The a0 register holds the pointer to top of new shadow
# to switch to. The current ssp is first pushed on the current
# shadow stack and the ssp is restored from new shadow stack
save_shadow_stack_pointer:
    ssprr  x5                   # read ssp and push value onto
    sspush x5                   # shadow stack. The [ssp] now
                                # holds ssp+8. Save away x5
                                # into a context structure to
                                # restore later.
restore_shadow_stack_pointer:
    ssamoswap t0, a0, x0        # t0=*[ssp] and *[ssp]=0
    addi      t0, t0, (XLEN/8)  # t0+XLEN/8 must match to a0
    bnez      t0, a0, crash     # if not crash program
    csrw      ssp, t0           # setup new ssp
```
There is an error on restore function. The `a0` register is expected to hold the memory address stored the checkpoint, the value should be `ssp`. Therefore, the value of `*[a0]` is expected to be `ssp+(XLEN/8)`, and the `t0` register is holding the value after `ssamoswap` instruction. In order to check the equivalent of the `t0` and `a0` register, you need to add `(XLEN+8)` to the `a0` register instead of the `t0` register. The correct equation would be: `t0 == a0+(XLEN/8)`. The `t0` register is still holding the top return address pointer, `ssp+(XLEN/8)`, after equivalent check.

The updated example is shown as following:
```
# The a0 register holds the pointer to top of new shadow
# to switch to. The current ssp is first pushed on the current
# shadow stack and the ssp is restored from new shadow stack
save_shadow_stack_pointer:
    ssprr  x5                   # read ssp and push value onto
    sspush x5                   # shadow stack. The [ssp] now
    addi   x5, x5, -(XLEN/8)    # holds ssp+(XLEN/8). The [x5]
                                # now holds ssp. Save away x5
                                # into a context structure to
                                # restore later.
restore_shadow_stack_pointer:
    ssamoswap t0, a0, x0        # t0=*[a0] and *[a0]=0
                                # The [a0] should hold spp
    addi      a0, a0, (XLEN/8)  # a0+XLEN/8 must match to t0
    bne       t0, a0, crash     # if not crash program
    csrw      ssp, t0           # setup new ssp
```

ved-rivos, 2023-02-05:
Thanks. The updated example looks good.

ved-rivos, 2023-02-05:
PR #32 to fix the example.

closed 2023-02-05 by ved-rivos