riscv/riscv-cfi #37: Binary distance between lpcll and lpsll issue for microarch/physical attacks?

kacouane, 2023-02-09:
I think this CFI extension might be the target for physical attacks.

There is a risk that a lpcll may be maliciously transformed into a lpsll.
Or a lpcml into a lpcll

With a bit of luck an attacker may change the label expected value with a glitched landing pad of the same value...

I know that there is example for such type attacks on other architectures.

I link a paper that has a nice introduction to the issue (just read the physical Attacks section)

https://past.date-conference.com/proceedings-archive/2019/pdf/1074.pdf

I'm not sure but I think that it the binary distance between these two critical instructions was higher the risk might be reduced.

Was this issues considered? Is there any mitigation planned?

G3org1o, 2023-02-09:
Hi, physical attacks are outside of the scope of this specification as we only try to protect the control-flow from being hijacked through memory corruption vulnerabilities (that can corrupt control flow variables in the process's address space). Trying to fit physical attacks, debugging protections, etc. to our threat would require SSLP to be supplemented
with various other mechanisms that provide solution for these cases. 

There are numerous academic proposals that try to protect systems from physical attacks that are orthogonal to our specification. For example https://www.cs.tau.ac.il/~mad/publications/asplos2016-iommu.pdf proposes the utilization of the IOMMU to prevent DMA attacks. 

Binary distance seems inadequate, for example researchers have proven that bit-flipping attacks can even target ECC RAM https://www.vusec.net/projects/eccploit/ (i.e. flipping more than one bit to alter the memory region targeted).

ved-rivos, 2023-02-17:
Closing as physical attacks are outside of scope.

closed 2023-02-17 by ved-rivos