riscv/riscv-cfi #47: Hidden Execution Path of Backward Edge CFI

henry-hsieh, 2023-02-21:
In [this](https://arxiv.org/abs/2103.08229) ROP attack paper, the authors managed to attack RISC-V using hidden execution path in the code. The return instructions return to the addresses which are not the instruction heads. This only happen when C extension is enabled. In this case, the attacker may find return instruction without the shadow stack protection to construct the gadgets.

ved-rivos, 2023-02-21:
Thanks. This paper shows ROP technique that relies on corruption of the return address variable to start a ROP chain. Software techniques like stack canaries can be defeated has been demonstrated. Transferring control to invalid instructions including starting execution in middle of an instruction to create these hidden instructions is also a well known technique with unaligned ISA like x86 and risc-v (with C extension) - was originally shown by [Shacham](https://hovav.net/ucsd/dist/geometry.pdf). Bypassing function prologues and hence checks as done in this paper by transferring to middle of the setuid function is also commonly used by code reuse attacks. 

These threats and techniques are addressed by the RISC-V CFI extension with hardware enforced shadow stack and labeled landing pads to protect the control flow to ensure that return address variables are protected and control transfer occurs only to valid targets (and not middle of instructions) in the program. Please also see issue #25.

henry-hsieh, 2023-02-22:
OK. Basically, the shadow stack is used to stop the starting return instruction of a ROP chain, which must be the main instruction instead of the hidden instruction, even though other gadgets may use hidden instructions. That why return instruction doesn't require ELP alike landing pad.

ved-rivos, 2023-02-22:
Yes, the shadow stack provides a protected store for the return addresses so that returns can only be to instruction after a previous call.

henry-hsieh, 2023-02-23:
Okay! I'll close the issue. Thanks for the help.

closed 2023-02-23 by henry-hsieh