riscv/riscv-cfi #48: Update needed after sspush change

jklockars, 2023-02-22:
Under "3.1. Backward-edge CFI instruction encoding", the text regarding "Encoding * as..." needs to be updated now that sspush is changed.
The text under the first table should probably be
  Encoding rd as x0 is not supported for sspop and ssprr.
and under the second table
  Encoding rd as x0 is not supported for ssamoswap. Encoding rs2 as x0 is not supported for sspush.

Also, it looks a bit confusing that rs1 is noted with contents "0" for sspop/ssprr/sspush and rs2 as "1" for sschkra, while rs1 for sschkra is "00101". Either have "5" in the latter case, or (better, IMHO) use "00000" and "00001" for the others.

ved-rivos, 2023-02-22:
Thanks. `sspush` does not need a restriction on source now. Updated with suggestion and corrections in PR #49

jklockars, 2023-02-22:
True, although since it does not make much sense to ever sspush x0, it would save a bit pattern for possible later use.

ved-rivos, 2023-02-23:
I am questioning if saving that one pattern is useful. There are 31 patterns available with `rd=0` and one more with `rd!=0` in `mop.r.2` ..

jklockars, 2023-02-23:
Allowing sspush x0 will shorten some logic ever so slightly, so I'm certainly not against it. Just thought it perhaps should be considered.

By the way, wouldn't sspush normally (just about always?) be done with x1 or x5? Enforcing that would save more opcode space.

closed 2023-03-17 by ved-rivos