riscv/riscv-cfi #54: Priority of forward CFI fault?

jklockars, 2023-03-06:
In my implementation of CFI, the checks need to be done late in the CPU pipeline. This means that a lot of other things have been checked before it is decided whether a call target has the expected lpcll.
For example, the target instruction could be illegal, a misaligned branch, a faulting memory access etc.

All such things will of course also cause a fault handler to be called, and some of them will eventually cause the instruction in question to be restarted - whereupon any CFI fault will be taken.
Still, for the cases where an instruction restart does _not_ happen (there is another fatal fault), do we want there to be some kind of fault prioritization, or does it not matter?

Whether prioritization is expected or not should (IMHO) be documented.

ved-rivos, 2023-03-07:
The priority would be as follows for the illegal instruction exception caused by lpcll.
JALR commits and moves ELP to LP_EXPECTED
1. Asynchronous events - timer, external int, SW int, local ints, debug, etc.
2. Instruction Page fault, Instruction guest page fault, Instruction access fault on fetch from the JALR target
<<now the instruction bytes are available>>
3. Illegal instruction if instruction is not aligned to 4 bytes, not a LPCLL, or if LPLCLL.LLPL does not match LPL.LL

All trap delivery (1, 2, and 3) move ELP to xPELP and set ELP to NO_LP_EXPECTED.

> Still, for the cases where an instruction restart does not happen (there is another fatal fault)

This restart does not happen because the trap handler decides to terminate the software context and not SRET to it?

jklockars, 2023-03-08:
> The priority would be as follows for the illegal instruction exception caused by lpcll. JALR commits and moves ELP to LP_EXPECTED
...

I am not talking about that prioritization, which of course needs to be as you say.

> > Still, for the cases where an instruction restart does not happen (there is another fatal fault)
> 
> This restart does not happen because the trap handler decides to terminate the software context and not SRET to it?

Yes. The target instruction could, for example, try to read from a non-aligned address (on hardware that does not support such).

The question is if we should enforce a priority in that case, so that the instruction must cause an illegal instruction fault (as a bad CFI target instruction) rather than a misaligned load access fault (or, for that matter, the other way around). Either would normally be unrecoverable ("fatal"), and it should not (AFAICT) normally make any difference which was reported.
I don't think there has been a similar situation in RISC-V before.

It is not as easy as to just say that the instruction should actually decode as illegal when CFI is active. For an SSCHKRA that is normally impossible (since it needs register contents to decide whether to "be illegal" or not), but depending on implementation it could also be difficult for LPCLL.
On the NOEL-V, the only reasonable place to deal with these instructions was in our "exception stage", stage 6 out of 7 (just before write-back). At that point, all other possible exceptions have already been checked for, so it would take extra effort if the "CFI illegality" should override (the example above is currently reported as a misaligned load, with xPELP set).

jklockars, 2023-03-08:
Replying to myself here.

> The question is if we should enforce a priority in that case, so that the instruction must cause an illegal instruction fault (as a bad CFI target instruction) rather than a misaligned load access [...]

As you point to in the PR, there is actually a defined order for these synchronous faults. I had forgotten about that because the order, until now, was pretty obvious.

> It is not as easy as to just say that the instruction should actually decode as illegal when CFI is active.
...

I will apparently have to deal with this, replacing some previously marked synchronous exceptions with the illegal instruction caused by forward-CFI. Hopefully it will not be too "interesting"...

ved-rivos, 2023-03-08:
Agree. I will apply the PR now and resolving these two issues.

closed 2023-03-09 by ved-rivos