riscv/riscv-cfi #74: Store to load forwarding

deanliberty5, 2023-03-30:
I found the following note a bit confusing:

"Store to load forwarding is a common technique employed by high performance
processor implementations. CFI implementations may restrict forwarding from a
non-shadow-stack store to a sspop instruction. A non-shadow-stack store causes a
fault if done to a page mapped as a shadow stack. However such determination
may be delayed till the PTE has been examined and thus may be used to
transiently forward the data from such stores to a sspop."

What exactly is meant by " may restrict forwarding"?

ved-rivos, 2023-03-30:
To not forward from a non-shadow-stack store to a sspop or sspopchk.
The scenario is as follows:
beq <somewhere> # this branch will mispredict
sd [x9], x2              # x9 contains shadow stack address and x2 malicious redirect target
                              # but this store is executing transiently, the beq will mispredict and flush it away
                              # if this was not speculative the sd would have faulted anyway since store to SS
                              # is disallowed.
sspop x1               # sspop executes speculatively and ssp matches [x1] so the store if it forwards to
                             # sspop then x1 now contains the malicious value in x2
c.jr                        # speculative execution may occur from [x2]
                             # eventually the branch will mispredict and flush all of this away.

ved-rivos, 2023-03-30:
Would changing restrict to prevent be more clear?

deanliberty5, 2023-03-30:
I see.  Yes, that explanation is helpful.  Changing "restrict" to "prevent" is probably a good idea.  Others may benefit from your scenario description, as well.  (If it's already obvious to everyone else, you don't need to bother.)

In the scenario you posed, even if store-to-forward is performed, changes to x1 should not become architecturally visible, should they?

ved-rivos, 2023-03-30:
The x1 would not be architecturally visible. I will update "restrict" to "prevent".

ved-rivos, 2023-03-30:
Updates in https://github.com/riscv/riscv-cfi/pull/73

closed 2023-03-30 by ved-rivos