riscv/riscv-cfi #79: setjmp/longjmp example (at end of section 3.2)

thurstond, 2023-04-04:
### 1. Implicit assumption that setjmp is a leaf function and doesn't spill to the shadow stack

The spec mentions that the compiler/author can choose not to use the shadow stack for leaf functions, but does not mandate it [1]. However, this optimization is assumed by the setjmp/longjmp example; if setjmp did spill to the shadow stack, there would be an off-by-one error in longjmp: it would be partially restoring the state of setjmp, not setjmp's caller.

Could this assumption be made explicit in the text?

[1]

> A leaf function i.e. a function that does not itself make function calls does not need to push the link register to the shadow stack or pop it from the shadow stack in either shadow stack mode or in control stack mode. The return value *may* be held in the link register itself for the duration of the leaf function execution.

(emphasis added)

There is somewhat stronger wording in the Introduction:

> The shadow stack is used to spill the link register if required by non-leaf functions.

but I read that as a suggestion of where the shadow stack could be used, not as an exclusion of other uses.

----

### 2. Editorial: confusing code comment for setjmp

```
  // read and save top of stack pointer to jmp_buf
  asm("ssprr %0" : "=r"(cur_ssp):);
  jmp_buf->saved_ssp = cur_ssp;
```

Should this comment be "read and save shadow stack pointer to jmp_buf"? I interpreted the comment to be referring to the code that immediately follows.

(The existing comment could be argued to be technically correct, because setjmp does typically save the (non-shadow) stack pointer, or thereabouts.)

----

### 3. Editorial: guard pages are mentioned in the context of longjmp/setjmp, which is partly out of context

The text currently reads:

> The unwind function must verify that it does not accidentally unwind past the bounds of the shadow stack. Shadow stacks are expected to be bounded on each end using guard pages i.e. pages that do not have a shadow stack attribute.

Unwinding justifies why a guard page is needed at the bottom of the shadow stack, but it does not explain why a guard page is needed at the top end of the shadow stack. I can imagine there are good reasons (not overflowing the shadow stack into user memory etc.), but it's not explained here (and the longjmp/setjmp section would not the proper place to do so). Perhaps the guard page requirement could be stated in section 3.5.1 (where the requirement to not alias writeable non-shadow-stack pages is listed)?

ved-rivos, 2023-04-04:
1. Agree. Will call that out in the text.
2. Agree. Will update as suggested
3. Overflow into user memory would not be a concern as it would lead to a fault. The non shadow stack page is itself a guard page. However If the overflow occurs to a unrelated shadow stack then that would be an error. Besides unwind, an underflow would also be a concern for the top end. Will expand that commentary here and also call out that the guard page while traditionally for a stack is a page inaccessible to the process, a guard page for shadow stack may be either a traditional guard page or could be a non-shadow stack page that cannot be used as a shadow stack but is otherwise accessible to the process.

thurstond, 2023-04-04:
Thanks for the quick reply! That's a good point - I had overlooked that about non-shadow stack pages.

ved-rivos, 2023-04-05:
Addressed in https://github.com/riscv/riscv-cfi/pull/81/commits

closed 2023-04-05 by ved-rivos