riscv/riscv-cfi #80: Nitpick: comments in sspop and sspopchk examples (Section 3.2)

thurstond, 2023-04-04:
```
  sspopchk x1 # compare link register x1 to shadow
             # return address; faults if not same
  #
  # sspop loads the value from location addressed by ssp and
  # compares the loaded value to the value held in the register
```
The code is sspopchk but the comment mentions 'sspop'.

-----

Conversely:
```
  sspop x1 # pop return address from shadow stack
  #
  # sspop loads the value from location addressed by ssp into
  # destination # register and updates ssp to (ssp + XLEN/8)
  # - does a pop. Following completion of sspopchk the ssp value
```

the code is sspop but the comment mentions 'sspopchk'. There's also an unnecessary '#' between 'destination # register'.

ved-rivos, 2023-04-05:
Addressed in https://github.com/riscv/riscv-cfi/pull/81/commits

closed 2023-04-05 by ved-rivos

thurstond, 2023-04-05:
Thanks! There's still one reference to sspopchk in the sspop example:
```
# **sspop** loads the value from location addressed by ssp into
# destination register and updates ssp to (ssp + XLEN/8)
# - does a pop. Following completion of **sspopchk** the ssp value
# is the new top of stack i.e. 0x19740428
```

----

On the topic of sspop vs. sspopchk, there's a few places where properties of sspop are mentioned, without mentioning sspopchk. Should they generally share the same properties?

Section 3.5.1:
- "The sspop instruction can only load from shadow stack memory."
- "When G-stage page tables are active, the ssamoswap, and sspop instructions require the G-stage page
table mapping the accessed memory to have read permission."

Section 3.5.2:
- "When privilege mode is less than M, ... the PMP region accessed by sspop must provide read permission."
- "The M-mode memory accesses by sspop instruction tests for read permission in the matching PMP
entry when permission checking is required."

Section 3.2:
- "CFI implementations may prevent forwarding from a non-shadow-stack store to a sspop instruction. ... However such determination may be delayed till the PTE has been examined and thus may be used to transiently forward the data from such stores to a sspop"

ved-rivos, 2023-04-05:
Thanks. Updated in https://github.com/riscv/riscv-cfi/pull/83/commits