riscv/riscv-cfi #82: Landing pad multi-labeling

thurstond, 2023-04-05:
###  Multi-labelling without call site specific entry points

In Chapter 1, a suggested use case is:

> For example, consider two call sites A and B. A invokes functions X and Y. B invokes functions Y and Z. With a
> single label scheme the functions X, Y, and Z would need to be assigned the same
> label such that both call site A and B can invoke the common function Y. This
> allows call site A to additionally call function Z and call site B to additionally call
> function X. However, if function Y was labeled with two labels - one corresponding
> to call site A and other to call site B then Y can be invoked by both, but the X can
> only be invoked by call site A and Z only by call site B. To support multiple labels,
> the compiler may create a call site specific entry-point to such shared functions
> with each entry-point containing the call site specific landing pad instruction
> followed by a direct branch to the start of the function.

It's possible to avoid the need for multiple call site specific entry points, by using the middle label:
```
Call site A:
- set lower label to 10
- set middle label to 20

Call site B:
- set lower label to 10
- [middle label is implicitly set by lpsll to 0]

X prologue:
- check lower label is 10
- check middle label is 20

Y prologue:
- check lower label is 10

Z prologue:
- check lower label is 10
- check middle label is 0
```
Perhaps this straightline code could be more efficient than the entrypoint trampolines?

### Using lower + upper label instead of lower + middle label?

The spec states:

> For label widths up to 17-bit a companion instruction lpcml is provided.
> ...
> A lpsml instruction is provided to set the value of the middle label (ML) field of the lpl. This
> instruction is used when labels wider than 9-bit are used.
>
> ...
> A lpsul instruction is provided to set the value of upper label (UL) field lpl. This instruction is used
when labels wider than 17-bit are used.

AFAICS it would be equally valid to use the lower+upper labels to handle the 17-bit case; i.e., it is legal to check the lower+upper labels without checking the middle label:
```
foo:
  lpsll $10
  lpsul $30 # Instead of lpsml
  jalr %ra, %x10

bar:
  lpcll $10
  lpcul $30 # Instead of lpcml
  : 
```

Is this allowed?

If so, the text is not quite accurate in stating that lpsml is used for >9-bit labels and lpsul is used for >17-bit labels.

Besides being a technicality, being able to check the upper label without checking the middle label gives additional multi-labelling (without call site specific entrypoint) flexibility e.g.,
```
Call site A:
- set lower label to 10
- set middle label to 20
- [high label is implicitly set to 0]

Call site B:
- set lower label to 10
- [middle label is implicitly set to 0]
- [high label is implicitly set to 0]

Call site C:
- set lower label to 10
- [middle label is implicitly set to 0]
- set high label to 30

Call site D:
- set lower label to 10
- set middle label to 20
- set high label to 30

X prologue:
- check lower label is 10
- check middle label is 20

Y prologue:
- check lower label is 10

Z prologue:
- check lower label is 10
- check middle label is 0

U prologue:
- check lower label is 10
- check high label is 30

V prologue:
- check lower label is 10
- check high label is 0
```
enables the CFI policy:
```
A -> {X, Y,       V}
B -> {   Y, Z,    V}
C -> {   Y, Z, U  }
D -> {X, Y,    U  }
```

Many more policies are possible by varying the combinations of label values.

### Wacky idea: split the label into more pieces

Currently, the 25-bits of the landing pad label are split into 9/8/8 low/mid/high bits. However, if it was split into more pieces (e.g., 7/6/6/6 bits, with four set/check landing pad instructions; or 5/5/5/5/5 with five instruction pairs, etc.), it will give even more flexibility to do multi-labelling without trampolines.

ved-rivos, 2023-04-05:
> Perhaps this straightline code could be more efficient than the entrypoint trampolines?

While this is possible, it may or may not be safe. For example, Y in this example checks for lower label 10. If the only call sites in the program with lower label of 10 were A and B this would be okay. However if there was a unrelated call site say C that also employed lower label of 10 then it would be able to invoke Y even if there was not a relation between C and Y. Enforcing this labeling may then require that a lower label of 10 cannot be used in any other relation which may cause an exhaustion of labels.

> AFAICS it would be equally valid to use the lower+upper labels to handle the 17-bit case; i.e., it is legal to check the lower+upper labels without checking the middle label:

Yes, it would be valid. The text was describing the canonical usage. A piece of text that the use of middle is not required can be inserted.

thurstond, 2023-04-05:
> While this is possible, it may or may not be safe. For example, Y in this example checks for lower label 10. If the only call sites in the program with lower label of 10 were A and B this would be [...]

I agree that, with the straightline code snippet, the lower label of 10 would have to be reserved for A and B, to prevent unrelated calls.

To avoid exhausting the number of lower labels, the scheme can be generalized to use a combination of lower and middle labels:

```
Call site A:
- set lower label to 10
- set middle label to 20
- [high label is implicitly set by lpsll to 0]

Call site B:
- set lower label to 10
- set middle label to 20
- set high label to 30

X prologue:
- check lower label is 10
- check middle label is 20
- check high label is 0

Y prologue:
- check lower label is 10
- check middle label is 20

Z prologue:
- check lower label is 10
- check middle label is 20
- check high label is 30
```

Here, the combination of lower label = 10 and middle label = 20 must be reserved; there are 2^(9+8) == 131072 different combinations of lower and middle labels.

Granted, this doesn't have much flexibility because the only remaining degree of freedom is the high label; this is a case where splitting the label into more pieces would be useful.

> Yes, it would be valid. The text was describing the canonical usage. A piece of text that the use of middle is not required can be inserted.

That would be helpful!

ved-rivos, 2023-04-06:
Thanks. I will add that note. Would be great if this scheme can be prototyped!

ved-rivos, 2023-04-07:
https://github.com/riscv/riscv-cfi/pull/85/commits adds the note.

closed 2023-04-10 by ved-rivos