riscv/riscv-cfi #84: Jalr as indirect jump/call question

kacouane, 2023-04-07:
Hello,

The determination of JALR as an indirect call/jump (that needs to target a lpcll) is not following the hints given by the [riscv-isa](https://github.com/riscv/riscv-isa-manual/releases/download/Ratified-IMAFDQC/riscv-spec-20191213.pdf) for RAS

[image]

In my opinion, the jalr that are included in the red square (1 for color blind) shall not need to target a landing pad
While the ones in the second shall require one as forward transitions (push in RAS).

I think that if we don't follow the same hints for this extension it may reduce its adoption.

ved-rivos, 2023-04-07:
The first row in box 1 is a indirect call - its just not using the conventional link register. It should require a landing pad.

The second row in box is is a return if rd is x0 - it will not require a landing pad - but if the rd is not x0 its an indirect call - its should require a landing pad.

kacouane, 2023-04-07:
The issue in my opinion is it's not an unconventional call it can be an unconventional ret or just anything you can imagine that might not need a landing pad.

I think about jump tables that will in fact need the same label anyway on every target ...

kacouane, 2023-04-07:
I think it might not be the right choice to say everything that is not a conventional ret shall have a landing pad

ved-rivos, 2023-04-07:
If its not a return, then its a indirect forward branch. The return is allowed to not require a landing pad because the return address variable is provided by hardware in the link register and is protected from tamper when spilled on the shadow stack. All other addresses are explicitly taken and when an address is taken it causes a landing pad to be produced at that address indicating that is the only valid location where branches can land since the address was taken by the program.

kacouane, 2023-04-11:
I can see many example were a landing pad is not necessary for security.

below two examples:

let say I create a function that uses an "unconventional ret" I can still use the shadow stack if I want to enforce the flow path. 
I'm not sure why it shall need a landing pad on the return address
```
bar:
    sspush x0
    sd x0 -4(sp)
    addi sp, sp, -37
    mv s1, sp
    j <scanf_unsecured>
   ...
   sspopchk x0
   jalr x10, x0
```

the second example is just the basic switch optimization by GCC that use a jump table.
For now the policy when using CET BTI, the solution is to disable the forward protection or to put on every target the same label when no address can be tampered and were compiler generated.
see discussion [here](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=104816)

Both these examples show that it might not add anything to request a lpcll after every jalr in the first two rows and, following the hints given for the RAS might help the adoption of Zisslpcfi extension.

ved-rivos, 2023-04-12:
I think I see the issue we have in the equation for the case 2 i.e. second row in the table and we should update as follows:
```

    if ( ((JALR && rd == x0) || C.JR) && (rs1 != x1) && (rs1 != x5) )
        is_indirect_jump = 1;
    else
        is_indirect_jump = 0;
    if ( ((JALR && rd != x0) || C.JALR ) && (rs1 != x1) && (rs1 != x5) )
        is_indirect_call = 1;
    else
        is_indirect_call = 0;
    is_indirect_call_jump = is_indirect_call | is_indirect_jump;

```
The first row is a indirect jump - so should require a landing pad. Shall we speak on a call why we would not want to protect that free branch - not protecting row 1 would mean we are not guarding against jump oriented programming techniques.

ved-rivos, 2023-04-14:
We may not  want to treat all of row 1 as unprotected. But we could provide for a software guarded branch. A software guarded branch would be a indirect branch instruction with a register destination that is not x0, x1, or x5. To use a software guarded branch, software should ensure that the rs1 used for the branch is either not mutable or has been verified to be valid through software means. For example, in case of a switch case, the list of targets may be in a read-only table and the compiler may add checks to ensure that the index into the table has been bounds checked. While software guarded branches may in most cases be secured, in some cases they may still be susceptible. For example, where software can be interrupted and there by the registers spilled, then there is a possibility of a TOCTOU issue where after the bounds check and address compute was done by SW, the interrupt or signal handler spilled the register holding the address or index to memory where it could be modified. When the signal/interrupt handler returns the modified address could be used without further guards.

ved-rivos, 2023-04-17:
Please review - https://github.com/riscv/riscv-cfi/pull/93/commits

closed 2023-04-18 by ved-rivos