riscv/riscv-cfi #89: Prerequisites and dependencies

deanliberty5, 2023-04-12:
I just realized that smepmp might be a prerequisite for SS and LP.  It is not stated explicitly.  Could you please include a list of perquisites and dependencies?

Thanks.

ved-rivos, 2023-04-12:
Priv: Smepmp and Sv39/Sv48/Sv57
Unpriv: Zicsr, Ziccif, Ziccamoa, A, and C

ved-rivos, 2023-04-12:
Updated extension dependencies list in PR #90 with latest updates (Zcmop, compressed cfi insts, Ziccamoa) and the missing list of priv extensions.

deanliberty5, 2023-04-12:
Ved, before we close this issue, I'd like to probe a little more.
**Why** is there a dependency on smepmp?  What is it in smepmp that we require?

ved-rivos, 2023-04-12:
Smepmp introduces mseccfg CSR. The Zisslpcfi adds a field in this CSR. Zisslpcfi can be implemented without Smepmp, however without Smepmp, the shadow stack memory protection is not present. 

> 
> When the `Smepmp` extension is implemented, a new WARL field `sspmp` is defined
> in the `mseccfg` CSR to identify a PMP entry as the shadow stack memory region
> for M-mode accesses.
> 
> When `sspmp` field is implemented and `mseccfg.MML` is 1 the following rules are
> additionally enforced for M-mode memory accesses:
> 
> * `sspush`, `c.sspush`, `sspop`, `sspopchk`, `c.sspopchk`, and `ssamoswap`
>   instructions must match PMP entry `sspmp`.
> 
> * Write by instructions other than `sspush`, `c.sspush`, and `ssamoswap` that
>   match PMP entry `sspmp` cause an access violation exception.

ved-rivos, 2023-04-12:
Though it is possible to make dependence only on the `mseccfg` CSR and not the rest of the `Smepmp`.

deanliberty5, 2023-04-12:
Yes, that is the dependency we would prefer to see.  

I am sitting here reading about MML, to make sure I have not overlooked anything.  We obviously need the encoding of pmpcfg.RW=01.

ved-rivos, 2023-04-12:
The `MML` if implemented would write protect `sspmp` field. The protection itself should not be dependent on `MML` - only whether `sspmp` field is implemented. See https://github.com/riscv/riscv-cfi/pull/90/commits

closed 2023-04-14 by ved-rivos