riscv/riscv-cfi #113: Landing pad encoding update

ved-rivos, 2023-06-10:
Sharing an proposal for alternate encoding of landing pads from members of AR (has not been discussed in AR yet). The `lpcll` instruction to be renamed as `lp` (landing pad) and the `lpcnl` removed. `lp` instruction embeds a 20-bit immediate as the label of the landing pad. This instruction would be encoded as `auipc` with `rd=0` - a code point that is presently a designated HINT.The encoding using the U-type hint provides a compact encoding for 20-bit labels. The indirect call/jump site would used `auipc x7, $label_imm` instruction to setup the expected landing pad label. The landing pad would be encoded as `lp $label_imm`.

ved-rivos, 2023-06-11:
PR #114 with proposed update

closed 2023-06-17 by ved-rivos

Larhzu, 2023-06-23:
I know this is a little late but I hope it's OK. I'm very much not an expert so if the following doesn't make sense, I apologize for the noise. :-)

1. The 20-bit landing pad label sounds nice and compact. However, it means that `lpad` has an implicit read from x7 which is unusual in RISC-V. Requiring that `lpad` must have x7 in inst[19:15] (auipc_rs1) would avoid this. The label becomes 15 bits though.

2. Concatenation of two instructions can form a landing pad: This can be made less likely or even impossible in common situations. (By the way, in cfi_forward.adoc `?013h` should be `?017h` to match `auipc`.)

As long as there are only 16-bit and 32-bit instructions, it's possible to define a landing pad that cannot be accidentally formed, even if the landing pad is only two-byte aligned. It is still possible to embed a landing pad label and encode the landing pad register in auipc_rs1.

The idea is that the highest 16 bits of the landing pad must be chosen so that compilers will never emit instructions that begin with those bits. For example, `auipc x0, 0` would be such if `c.unimp` (0x0000) was never emitted: even though some 32-bit instructions end with 0x0017, an accidental landing pad cannot form if `c.unimp` is never used (especially valuable for the unlabeled landing pad). Currently Clang/LLVM can emit `c.unimp` in unreachable code (e.g. as padding immediately after `jal`) but GCC doesn't, I think.

For a 6-bit label and auipc_rs1=x28/t3 or x29/t4:

Eight 16-bit hints (`c.mv x0, x3`, `c.mv x0, x7`, ...) could be reserved for use as the high 16 bits of a landing pad. Compilers would never emit these hints so accidental landing pads wouldn't be possible as long as instructions longer than 32 bits aren't present. The eight hints provide a 3-bit label and another three bits are in auipc's [14:12]. But I guess a 6-bit label is unacceptably small.

For a 12/13/14-bit label and auipc_rs1=x30/t5 or x31/t6:

The unprivileged ISA spec has a table showing that instructions with the lowest seven bits set (0x7F) could be used for very long instructions. It's not frozen so it might change. If one is willing to guess that 0x7F prefix is unlikely to be used for popular instructions, it could be interesting for landing pad use instead of reserving 16-bit hints.

```
lui     x31, 0xnnvFw  # nn={00,...,FF}, v={7,F}, w={8,...,F}
                      # (If x30 is better then w={0,...,7}.)
jalr    rs
...
auipc   x0, 0xnnvFw   # Landing pad, compares bits [31:12] in x31.
                      # (Or sign-extend and compare [XLEN:12]?)

|31        23 | 22 20 | 19 16 | 15 | 14 12 | 11  7 | 6     0 |
|    label    |  111  |     rs1    | label |  rd   |  AUIPC  |
| nnnn nnnn v |  111  |  1111   1  |  www  | 00000 | 0010111 |
|                imm[31:12]                |
```

Currently no standard 32-bit instructions have the five lowest bits all set (0x1F is non-frozenly reserved for 48-bit instructions). So even 11 + 3 = 14-bit label (inst[31:21] + inst[14:12]) would fit with this scheme and accidental landing pads would be impossible when only currently-standard 16/32-bit instructions are present. It's not so future-proof in terms of accidental landing pads but perhaps the larger label is overall more valuable.

Comparable to the removed `lpcnl`, it is possible add 12 extra bits with `addi x31, x31, $label_low` and verify it with another hint instruction like `xori x0, x31, $label_low` but the extra code size isn't nice.

Avoiding accidental landing pads is most important with the unlabeled landing pad. All zeros (apart from [22:15] or [20:15]) or ones are the simplest choices but perhaps other patterns are worth considering too. For example, 0x707F at the high bits would be like a beginning of an instruction with opcode=0x7F, rd=x0, funct3=7. But it's impossible to guess what kind of bit patterns will be common inside long instructions in the future.

sorear, 2023-06-24:
Good catch on `?017h`.

I don't think it's a big problem to require alignment of landing pads, as almost all reasonable microarchitectures benefit from `-falign-functions=4` and it provides a universal method of avoiding accidental landing pads.

For clarity in the sequel: Define an _accidental landing pad_ as a pair of parcels which are a valid (possibly except for alignment) `lpad` hint but do not correspond to a `lpad` hint in the assembler input or equivalent (e.g. MIR).  The global property we need to establish is that no accidental landing pads are aligned.

Define a _hazardous pair_ as two consecutive instructions at the assembly or MIR level whose binary representation contains an accidental landing pad, and at least one parcel of the accidental landing pad comes from the first instruction.  Due to the prefix-free property of the RISC-V encoding, the accidental landing pad cannot start at parcel 0 of the first instruction, and since the only instructions currently supported have 1 or 2 parcels, the accidental landing pad must start at parcel 1.  As such, we can satisfy the global property by forcing all hazardous pairs to be aligned, either by adding a `c.nop` or by decompressing the most recently compressed instruction.

Define a _hazardous instruction_ as an instruction which can be the first instruction of a hazardous pair.  Hazardous instructions satisfy the bit pattern `????_0000_0001_0111_????_????_????_????`.  The following instructions are hazardous, and therefore must be aligned:

1. All `R` type instructions with `funct7[2:0]` = `0`, `rs2` = `x1/ra` or `f1/ft1`, `rs1` = `x14/a4` or `f14/fa4` or `x15/a5` or `f15/fa5`.

    `a4` and `a5` are problematic due to their frequently required use with the C extension, but uses of `ra` other than those mandated by the calling convention are rare, so this group of instructions should also be quite rare in practice, especially if the compiler is modified to move `ft1` to the end of the FP allocation order.

2. All `R4` type instructions with `funct2` = 0 (single precision), `rs3` even, `rs2` = `f1/ft1`, `rs1` = `f14/fa4` or `f15/fa5`.

    The same argument applies as for the `R` case with the register allocation change.

3. All `I` type instructions with `imm` = 1 + 256*n and `rs1` = `x14/a4`, `x15/a5`, `f14/fa4`, `f15/fa5`.

    This is a more problematic group, especially since addition of 1 and loading bytes at offset 1 are relatively common operations.  This also applies to `fcvt.wu.s`, `fcvt.s.wu`, `fcvt.s.d`, and `fli.s`, for literals 0.75 and 0.875 or `fa4`/`fa5` input, as well as writes to `fflags` using `a4` or `a5` as a source.

    `sll(w) RD, RS1, 1` can be rewritten as `add(w) RD, RS1, RS1` with no impact on latency or execution unit choice on typical microarchitectures.

    There is a potential problem if the instruction is subject to a relocation and therefore the immediate value is not known at assembly time.  Any instruction with a LO12_I-family relocation and a register of `a4` or `a5` is hazardous and must be aligned.  However, all LO12 style relocations either use a base from `tp`, `gp`, or `zero`, or generated by an `auipc`, `lui`, or `c.lui`, all of which have 5-bit register specifiers, so there is no particular difficulty in ensuring that the compiler never uses `a4` or `a5` as the intermediate register of a relocated pair.  (This may require new pseudoinstruction forms for `l*` with an explicit temporary.)

4. All `S` type instructions with `imm[7:5]` = 0, `rs2` = `x1/ra` or `f1/ft1`, `rs1` = `x14/a4` or `x15/a5`.

    Storing the return address or `ft1` anywhere other than the stack is not expected to be common; compare-and-branch on the return address even less so.

5. `J`-type instructions with `imm[7:1]` = 0, `imm[11]` = 1, and `imm[19:16]` = 7.  Superficially complicated, but since only an unaligned (`pc[1] = 1`) instruction can be hazardous, and only an instruction with `imm[1] = 0` can be hazardous, the only hazardous `jal` instructions are those to a destination which is 2 mod 4.

    If "all code symbols must be at addresses aligned to a multiple of 4" (which will be true of most functions due to the landing pads at their beginning) is elevated to a requirement of the landing pad ABI, then `jal` with a symbol as its target is never hazardous.  For intra-function `jal` instructions, the compiler has a choice between aligning the target and aligning the jump instruction, although this is only necessary if the function is larger than 448KiB due to the constraint on `imm[19:16]`.

6. U-type instructions with `imm` = `?017?h`.  Direct use of constants which require `lui` in this range (e.g. 1550000) is expected to be rare, however this is the most impactful case since it affects all instructions with a HI20-type relocation.

    Relaxation does not, in general, help here, since relaxation passes change symbol addresses and can convert previously-safe instructions with HI20 relocations into hazardous ones.  However, if certain global conditions are met, then all instructions with HI20 relocations are and will remain non-hazardous and the alignment can be removed, either by a relaxation pass or through the use of a new code model.

    For absolute and non-PIC, non-`medany` addressing, the condition is that no part of any symbol is at an address of `?017_????h` or `?016_F[8-F]??`; this limits a contiguous text or data segment to slightly less than 256MiB.  With the default TEXT_START_ADDR = 0x10000 the combined te [... cut ...]  of label are cost.  All instructions which are hazardous in Proposal 3, with the addition of `jal` instructions covering more than +448KiB or less than -512KiB regardless of alignment, must be immediately followed in sequence by a LOAD, LOAD-FP, STORE, STORE-FP, OP-IMM, JALR, or c.nop.  NO linker support is needed except to limit the range of function call relaxation and check the maximum range of HI20-type relocs not immediately followed by a LO12-type (only relevant with `-mexplicit-relocs`).

ved-rivos, 2023-06-24:
Thanks for catching `?017h`. 

The use of implicit source was done after much deliberation as the benefit of having 1 million labels vs. 32K labels is substantial and important even for modest sized programs and especially for large programs like the linux kernel, chomium, etc. The most common and portable labeling scheme takes the prototype of the function (e.g. the string "void (*)(int)") and employ a hash function (e.g., md5/sha1) and then use a truncated 20-bit value as a label. This allows linking object files that may have been differently compiled and still agree on the label use. With C extension there are cases of implicit source (e.g., c.lwsp), with vector extension the masked instructions using `v0`, for shadow stack instructions the `ssp` is an implicit source or destination for all of the shadow stack instruction. This choice was also discussed in the architecture review committee and with all of these considerations the use of the `x7` as the implicit source for `lpad` is deemed acceptable.

> (distros will be forced to choose between RVA20 support and providing system libraries with shadow stack support).

Yes, the ISA is still growing and this may make portable backward compatibility restricted for a bit. For example, use of Zbb which is mandatory in RVA22 would preclude running on machines with RVA20. While support for the CFI extension will likely be optional in the first profile where it shows up the `MOP` are expected to be mandatory in that profile version. So this should allow profiles compliant software after the introduction of `MOP` to be backward compatible - unless there are other restrictions such as Zbb.

Agree that the `lpad` requiring a 32-bit aligned `PC` allows for a set of cost neutral rules to be implemented in the compilers as illustrated by @sorear. Changing the landing pad encoding did not significantly change that property. When a sequence that may lead to an unexpected landing pad arises the compiler may emit a `c.nop` for example to make it unaligned or not a landing pad pattern. For many small and embedded programs the text may also be quite small and the number of needed labels be much lower.

sorear, 2023-06-25:
Is there a publicly available gnu-toolchain or llvm fork with landing pad support?  I'd like to generate hard data on the code size and linker runtime impact of accidental landing pads but I'm hesitant to duplicate more work than necessary.

So the label size is a driving factor.  Would it be fair to say that the landing pad was changed from a MOP to a HINT not because it's more suited to being a HINT than sspush/sspopchk are, but primarily in order to use the larger 20-bit label space?

To summarize my fears surrounding profiles and landing pad adoption: I expect MOP to be mandatory in RVA23.  I also expect V to be mandatory in RVA23.  However, V is a horrendously complicated extension (to the best of my knowledge, there is _no_ complete open hardware implementation of V 1.0 even now, 18 months post-ratification), and there is a large segment of commercial and academic hardware for which implementing V is not justifiable for NRE or (less often) area reasons.  If such hardware is designed according to the profiles spec, it is likely to target RVA22, which does not list MOP even as an optional extension, so unless the designers think about this problem specifically, it is unlikely to support MOP even though it would be easy to do so.  Meanwhile, binary software distributions are under more pressure to chase the latest security hardening features (which are not always distinguished from security bugfixes) than they are to chase the latest performance enhancements; Intel, Arm, and IBM's backward-edge CFI solutions are compatible with pre-CFI architecture subsets for normal functions, so distributions may adopt a policy of backward-edge CFI hardening, but ours is _not_.

Larhzu, 2023-06-25:
Thank you for the very exhaustive replies! I learned quite a few things.

It's great that implicit read from x7 is OK in this kind of case.

I agree that requiring 4-byte alignment for landing pads is good. Even if hazardous pairs were made impossible with 16/32-bit instructions, accidental landing pads could still exist inside longer instructions and then alignment can help. The case of J-type instructions is another a good reason.

My motivation for posting was that the implications of "assembler/linker can force instruction to be aligned to a 4-byte boundary" in the spec sounded pretty complicated and I wondered if these requirements could be simplified easily enough but I understand that the cost of the simplest solutions is too high.

In addition to the requirements for toolchains, a minor extra thing is that assembly programmers could need to know about it (warnings can help). It's not intuitive that `srai a0, a5, 1` needs to be 4-byte aligned. If optimizing for size, assembler (or even linker) could need to be smart enough so that it won't insert a `c.nop` when `c.mv a0, a5; c.srai a0, 1` would be shorter (at the expense of a longer dependency chain).

New extensions add more accidental landing pads. For example, Zbb has a few good candidates with rs1=a4/a5: `ctz/ctzw`, and with certain immediates `rori`, `roriw`, `bclri`, `bexti`, `binvi`, `bseti`.

I mention two more probably bad ideas for completeness (not proposals):

  - Requiring inst[17:16]=3 (18-bit label) would ensure that accidental landing pads cannot form with 16-bit instructions. Thus jumping into such a landing pad wouldn't immediately synchronize with the instruction stream. I haven't looked at what kind of effect this would have in practice, like how often the instruction after an accidental landing pad is valid.

  - Landing pads with inst[20:16]=0x1F cannot form accidentally with current and (likely) any future standard 16/32-bit instructions. If certain function signatures are especially common for indirect calls then it might help if the hash of popular functions had this kind of values. I guess it's not worth it but I mention this just in case.

Another way to look at accidental landing pads (I see @sorear edited his message so this is now partially duplicate):

1. If all landing pads are labeled and thus all `jalr`, `lui+jalr`, and `auipc+jalr` uses are prefixed with `lui x7` then `lpad` with full 20-bit label is good at the ISA level. Some problematic labels could be banned in the ABI, for example, with @sorear's proposed methods. Then those labels appearing accidentally in the code wouldn't matter. The unlabeled `lpad` is a special case but `auipc x0, 0` seems pretty good since only `c.unimp` can create an accidental unlabeled landing pad with 16/32-bit instructions. This property of `c.unimp` could be good to explain in the specification, even if in practice it might not matter if `-falign-jumps=4` is used because then `jal+c.unimp` cannot be hazardous.

2. If some (or all) code is built with unlabeled landing pads, then accidental labeled landing pads are a problem if attacker can control both the value of x7 and the function pointer. Here ISA restrictions on the possible landing pad labels could help to reduce the number of accidental landing pads (for example, @sorear's proposal 1 could be good as it's both quite effective and simple).

Here are a few commands to count instructions that create accidental landing pads in a particular file. `objdump` is the slowest part so a tempoary file can be a good idea.

```
riscv64-elf-objdump -d -M no-aliases FILENAME \
    | grep '[26ae]:' \
    | grep -E '\b[0-9a-f]017[0-9a-f]{4}\b' \
    | tr -s ' ' | cut -f3 \
    | sort | uniq -c | sort -nr

# Proposal 1, the only difference is 0->1 in "\b[1-9a-f]":
riscv64-elf-objdump -d -M no-aliases FILENAME \
    | grep '[26ae]:' \
    | grep -E '\b[1-9a-f]017[0-9a-f]{4}\b' \
    | tr -s ' ' | cut -f3 \
    | sort | uniq -c | sort -nr

# Proposals 1 and 2 combined:
riscv64-elf-objdump -d -M no-aliases FILENAME \
    | grep '[26ae]:' \
    | sed -rn '
        /\b[1-9a-f]017[0-9a-f]{4}\b/{
            N
            /\n.*\b([0-9a-f]{4})?[0-9a-f]{2}([08]3|[08]7|[19]3|[2a]3|[2a]7|[6e]7)\b/!P
            D
        }' \
    | tr -s ' ' | cut -f3 \
    | sort | uniq -c | sort -nr
```

Below is sample output of the first command from libQt5WebKit.so.5.212.0 from Debian. .text is 19,189,602 bytes.

```
   4297 slliw
   1455 addiw
   1258 addi
   1247 slli
    778 andi
    745 ori
    508 sltiu
    457 auipc
    310 srliw
    285 lbu
    281 fcvt.s.d
    176 csrrw       # csrrw zero,fflags,a4 or a5
     91 xori
     80 srli
[... 35 lines cut ...]
```

Proposals 1 and 2 combined:

```
    241 fcvt.s.d
     93 addi
     38 sraiw
     35 fcvt.s.wu
     25 srai
     18 fcvt.wu.s
      4 fmul.s
      4 fle.s
      3 jal
      1 lbu
      1 feq.s
```

About MOPs:

While I understand that maybe-ops can be convenient in general, I didn't understand why they are useful for shadow stack when reading the spec. I got an impression that `sspush`, `sspopchk`, and their RVC counterparts could be RVI/RVC hints as they don't write to any general purpose registers and the shadow stack memory is not available for normal memory access. So those instructions have a kind of invisible effect just like `lpad` has.

Excluding the control stack mode, the rest of the shadow stack instructions are needed in special locations only and thus they could be normal instructions that would be guarded with runtime detection (like `__attribute__(ifunc(...))`. Maybe they wouldn't even need to do the xBCFIE check in hardware.

However, my understanding of this topic is very inadequate and I might be missing several elephants in the room. So it's better that I won't comment MOPs or shadow stack instructions any further. :-)

ved-rivos, 2023-06-25:
Thanks. For assembly programmed functions this could be a a warning. For most of these cases the neutral transform outlined by sorear could be used to address. Even if all of these were caused to be misaligned by inserting a `c.nop` then for this example it adds 24 KiB for a 19 MiB binary which should be acceptable. 

The `sspush` and `sspopchk` would be emitted in each non-leaf function prologue and epilogue to push the link register to the shadow stack at function entry and to check the link register value against the shadow copy prior to returning from the function. So these may not be considered as special locations and it may not be desirable to insert ifunc calls in all such non-leaf functions.

Larhzu, 2023-06-25:
I agree that `sspush` and `sspopchk` cannot be behind ifunc.

In my previous message I tried to say that perhaps `sspush` and `sspopchk` could be RVI hints, and the compressed version could be RVC hints. For example, these should be available:

  - `add x0, x7, x1` = `sspush x1`
  - `add x0, x7, x5` = `sspush x5`
  - `add x0, x6, x1` = `sspopchk x1`
  - `add x0, x6, x5` = `sspopchk x5`
  - `c.add x0, x1` = `c.sspush x1`
  - `c.mv x0, x1` = `c.sspopchk x1`

Stores have the source register in rs2, including the compressed instructions. This way the compressed push and pop could both use x1, if it is preferred to avoid implicit reads from general purpose registers (or at least it makes things pretty).

Pushes and popchecks differ in one bit only:

  - `sspush` has inst[15]=1 and `sspopchk` has inst[15]=0.
  - `c.sspush x1` has inst[12]=1 and `c.sspopchk x1` has inst[12]=0.

When `xBCFIE = 0` the above instructions would revert to their RVI or RVC behavior. That is, they affect architecturally visible state just like `nop` does.

`ssload x1`, `ssload x5`, `sspinc`, `ssprr`, and `ssamoswap` would be regular new instructions that result in illegal-instruction exception on processors that don't support Zicfisslp. Zimop or Zcmop wouldn't be used. These instructions would be wrapped with ifunc or similar method. I don't know if these need different behavior based on `xBCFIE` or not.

I repeat that I don't have a good understanding about this topic. There might be more than one huge reason why my above text is a useless idea. But I hope that this message clarifies what I tried to say in my previous message.

sorear, 2023-06-25:
@Larhzu 

> In addition to the requirements for toolchains, a minor extra thing is that assembly programmers could need to know about it (warnings can help). It's not intuitive that `srai a0, a5, 1` needs to be [...]

My expectation is that gas and the llvm integrated assembler would automatically align hazardous instructions when in avoid-landing-pads mode.  You would only be able to shoot yourself in the foot if you are using directives to generate raw instruction bytes (e.g. `.word 0x00171017`; `.insn` does not have this problem because it always describes a single instruction of a single length which can be aligned).  Instructions would not be rewritten by the assembler into different instructions, only aligned.

> New extensions add more accidental landing pads. For example, Zbb has a few good candidates with rs1=a4/a5: `ctz`/`ctzw`, and with certain immediates `rori`, `roriw`, `bclri`, `bexti`, `binvi`, `bseti`.

This is largely why I phrased the reply in terms of R-type/I-type instructions instead of trying to give an exhaustive list.

> If some (or all) code is built with unlabeled landing pads, then accidental labeled landing pads are a problem if attacker can control both the value of x7 and the function pointer.

I'm unsure if this should be part of the threat model.

> Here are a few commands to count instructions that create accidental landing pads in a particular file.

I like your basic approach and have reproduced your results.  That roughly matches my assumptions for the minor instructions, although it might be useful to look at float-heavier code.

Because you are analyzing the final binary you are are substantially undercounting potentially hazardous instructions with relocations - since the decision to align them or not must be made before the addresses are known, and thus before it can be definitively known if they will be hazardous.

libQt5WebKit.so.5.212.0 has 431893 auipc instructions.  Every single one of them, except for the one in the PLT header, is paired with a jalr, addi, load, or store instruction; I will ignore the possibility of hazardous instructions with LO12 relocations since they will be aligned if the paired auipc is.

    grep auipc qt5webkit.s.noa | wc -l
    grep -A1 auipc qt5webkit.s.noa | grep -v '^--' | grep -v auipc |
        tr -s ' ' | cut -f 3 | sort | uniq -c | sort -nr

Of these, 255713 are within the stable non-hazardous offset range (243331 negative, 12382 positive) and can have their alignment deleted by linker relaxation; 176180 auipc instructions must be treated as hazardous even after relaxation, as well as 23503 non-auipc instructions.

    grep auipc qt5webkit.s.noa | sed 's/.*,//' |
        perl -nle '$_=hex($_); print $_<0x00170?"SPOS":$_<0x80000?"LPOS":
                   $_<0xf0170?"LNEG":"SNEG"' | sort | uniq -c
    grep -E ':\s*[0-9a-f]017[0-9a-f]{4}' qt5webkit.s.noa | grep -v auipc | wc -l

2831 auipc instructions are contributed by the PLT, this is less than 2% of the total and will be ignored for the math.

    <qt5webkit.s.noa perl -nle 'next unless /auipc/ && /^\s+(\w+):/; print if hex($1)<0x4adf30'

143127 of the forward auipc instructions are used to generate references into the rodata and later segments, leaving 45434 forward auipc instructions referencing the text segment.

    objdump -h libQt5WebKit.so.5 | grep -A1 rodata
    grep -vE 'auipc.*,0x[8-9a-f][0-9a-f]{4}' qt5webkit.s.noa | grep -A1 auipc |
        perl -nle 'next unless / # ([0-9a-f]+) / && hex($1) >= 0x16faea8;
                   if (/<(.*?)[+>@]/) { print $1 }' | wc -l 

Assuming that alignment is applied using a special "align hazardous instruction" relocation that is applied after layout and relaxation and each alignment results in an average of 1 byte of padding, (176180+23505)*1 = 195 KiB of nops need to be added to align hazardous instructions, 88% of which is accounted for by conservative handling of auipc.  Adding padding between the text and rodata segment reduces this to (45434+23505)*1 = 67 KiB of nops, plus a 4 KiB page table.  WITHOUT linker support, if the assembler simply generates a `.balign 4` before each potentially hazardous instruction, (431893+23505)*1 = 445 KiB of nops must be added.

@ved-rivos

> Even if all of these were caused to be misaligned by inserting a `c.nop` then for this example it adds 24 KiB for a 19 MiB binary which should be acceptable.

In the most conservative case (all 2^20-1 labels are allowed, support added only to the assembler with an unmodified linker) you are looking at 445 KiB for a 19 MiB binary.  Is that still acceptable?  Is 71 KiB (67 KiB nops, 4 KiB page tables) acceptable in combination with requiring an updated linker and a potentially controversial change to the memory map?

1.5 MiB of padding limits for DSOs larger than 1.5 MiB gives you 1000 maximum DSOs instead of 2000 for RV32 with a 3GiB user limit and a worst-case size distribution.  Is that acceptable?

libQt5WebKit.so.5.212.0 has 79883 rets, 30846 c.jalr, and a roughly 1:2 ratio of negative to positive stack adjustments; assuming 2 rets per function, estimate 39941 functions and 156KiB for the landing pads themselves.   If calls are labeled, this estimate increases to 276 KiB.  This dominates the cost of alignment with linker support, even more so if restrictions are placed on labels, but without linker support and the memory map change alignment costs more than the landing pads themselves.

@ved-rivos @Larhzu I agree that placing `sspush`, `sspopchk` behind runtime tests makes no sense at all.  Zimop is a genuinely useful new primitive, unfortunate as it is that it is likely to share a debut profile with V, and I have no preference between MOPs and normal R-type instructions for the CFI instructions other than `sspush`, `sspopchk`, `lpad`.

ved-rivos, 2023-06-25:
The initial idea was to use the RVI/RVC hints for this extension but that was not considered acceptable by the architecture review committee as hint are not allowed to modify state which the `sspush` and `sspopchk` would do by updating the `ssp`. About the concern about implementations needing to also bring V extension, there is a RVB profile being defined that is for such implementations that do not intend to pay for the V extension. In the RVB profile, the V extension is optional.

Only address taken functions need be labeled. So it is expected that we would also have a pass to prune out the functions that do not have address taken. I do expect that we would make use of the linker technique you outlined and need to pad otherwise. Together with this estimate the growth would be in the acceptable limit of less than 5% preferably and not exceeding 10%.

sorear, 2023-06-25:
Taking a trap is clearly a change to architectural state, so if "all hints must be indistinguishable from nop except by timing" is still a rule, `lpad` can't be a hint either.  If `lpad` is allowed to be a hint because it will never have a visible effect in a correctly-functioning program, then I don't see why the same logic doesn't apply to `sspush` and `sspopchk`.

ved-rivos, 2023-06-25:
The reasoning is [here](https://lists.riscv.org/g/tech-unprivileged/message/505). Whereas it is possible to mix application and libraries where only some of them have been compiled with the shadow stack instructions it is not possible to do that mix with landing pads. Since there is a single mode bit xFCFIE that controls the behavior for the entire application this was acceptable. There is concern that the RVI/RVC hints may have been embedded in existing binaries and changing their behavior to non-hint may cause unexpected results.

sorear, 2023-06-26:
I am aware of that message, it just doesn't make sense to me.  An ELF object can have one of three dispositions to backward-edge CFI:
* The object doesn't interact with the shadow stack and doesn't mess up the relationship between the shadow stack and the real stack (no shadow stack instructions, all function calls are properly nested except by calling system functions like longjmp and __cxa_throw)
* The object has shadow stack instructions mirroring all of its operations on the real stack.
* The object is unaware of the shadow stack, and either misuses hints (not currently applicable) or performs operations on the real stack without reflecting them on the shadow stack (shadow stack unaware implementations of unwinding, stack switching, etc).

If you link any number of objects in the first two categories, you can have UBCFIE=1 and maintain a shadow stack (which may be missing some frames, due to objects in the first category), but once you link an object of the third category UBCFIE must become 0.  The only draft ABI I could find for GNU_PROPERTY_X86_FEATURE_1_{IBT,SHSTK} uses identical enable/disable logic for each of them...

Whether the GNU x86 approach (opting individual object files into shadow stack support on the basis of a command line option, without any inspection or opt-in of the inline assembly that would actually cause problems) actually makes sense from a user perspective is a different question, but both forward and backward CFI have both local and nonlocal effects on the ABI, need a global enable, have a global enable, and act like nops when not enabled. 

incidentally: It is not possible to handle unwinding or stack swapping in an UBCFIE-agnostic and Zisslpcfi-agnostic fashion, because the instructions to read and write `ssp` are not MOPs; assuming that no MOPs are added for this, libc needs to maintain a variable shadowing UBCFIE that can be read by the implementation of longjmp, and it probably needs to be exported as a thread local or via a function for use by user-written stack switching routines.

---

There is an unrelated problem with the use of `lpad` to synthesize labels longer than 20 bits.   Suppose we are trying to support a 32-bit label:

    lpad 0x12345
    slli t2,t2,20
    lpad 0x67800

Except, oops, _both_ `lpad` instructions are valid landing pads.  If your threat model includes control transfer to an attacker-controlled function pointer with a random value in `t2`, you now have twice as many targets.

Proposed partial fix:

```
If xFCFIE != 0
    // If PC not 4-byte aligned then illegal-instruction
    if (pc[1:0] != 0 && ELP == LP_EXPECTED)
        Cause illegal-instruction exception
    // If landing pad label not matched -> illegal-instruction
    else if (inst.LPL != x7[31:12] && inst.LPL != 0)
        Cause illegal-instruction exception
    else
        ELP = NO_LP_EXPECTED
else
    no-op
endif
```

Now if your header uses `c.slli` only the first `lpad` is a valid landing pad.  Partial fix because it only addresses the problem if the C extension is present.

ved-rivos, 2023-06-26:
The support is for landing pad labels of 20-bit width. Wider labels may be supported by a future extension when wider immediate are introduced.

Larhzu, 2023-06-26:
@ved-rivos:
> The initial idea was to use the RVI/RVC hints for this extension but that was not considered acceptable by the architecture review committee as hint are not allowed to modify state which the sspush and sspopchk would do by updating the ssp.

Thank you! This was a detail that I hadn't understood. :-)

I think it would be valuable if the spec had a sentence explaining it near "The 32-bit instructions are encoded using the `SYSTEM` major opcode". For example: "RVI or RVC hints were not used for these instructions because they modify architecturally visible state, for example, `sspush` and `sspopchk` modify `ssp` whose value can be read with `ssprr`."

Similarly, cfi_forward.adoc could have something like: "A hint instruction can be used as `lpad` because in any correctly-functioning program it is equivalent to a `nop`. The `ELP` state can only be observed if `lpad` raises an illegal-instruction exception."

That being said, @sorear explains it well why mixing shadow-stack-unaware code isn't guaranteed to work even with the current spec. In practice it often would but so it would with `sspush` and `sspopchk` being hints.

@sorear:
> > If some (or all) code is built with unlabeled landing pads, then accidental labeled landing pads are a problem if attacker can control both the value of x7 and the function pointer.
>
> I'm unsure if this should be part of the threat model.

Maybe not. Building with unlabeled landing pads is inherently less strict than labeled anyway.

> libQt5WebKit.so.5.212.0 [...] estimate 39941 functions and 156KiB for the landing pads themselves.

For rough comparison: Debian's equivalent x86-64 package has .text size of 30,834,702 bytes but doesn't use landing pads. On Arch Linux, qt5-webkit 5.212.0alpha4-18 has .text size of 31,637,939 bytes and 84,516 `endbr64` landing pads (338,064 bytes).

The Debian packages clang-tools-16 and llvm-16 contain a small amount (under 1 %) of `auipc` uses where the consuming instruction isn't immediately after `auipc` or there may be more than one consumer (like read-modify-write with single `auipc`). So it's like explicit relocs was the default. I wonder how common such code will be in the future, although I understood that with linker support for hazardous instructions it wouldn't be a significant problem. (Off-topic: Knowing this would be valuable to me since it affects handling of RISC-V executables in data compression. Fused `auipc` + `jalr`, `addi`, or I/Zfh/F/D/Q load/store pairs are simple to handle but if a significant amount (5-10 % or more) will be unfused in the future it may need to be taken into account.)

I think I don't have the skills to comment these topics much more. I appreciate the considerate and educating replies from both of you, I have learned quite a few things. Thank you!