riscv/riscv-cfi #118: Does the pseudo-instruction CALL require LPAD?

Larhzu, 2023-06-16:
The expansion of `call` uses `jalr` with rs1=x1. After [the commit 157ebd](https://github.com/riscv/riscv-cfi/commit/157ebdf42e3d3fbad87c4636fa15b459583a1715) this seems to no longer require a landing pad at the target address. I wonder if this is a bug.

ved-rivos, 2023-06-16:
The expansion of `call` should be using `rd=x1` i.e. the `x1` holds the return address as `x1` is designated by the calling convention as the link register to hold the return address.  A `ret` should expand to `JALR`/`C.JR` with `rs1=x1`.

Larhzu, 2023-06-16:
According to the RISC-V Unprivileged ISA V20191213 or [RISC-V Assembly Programmer's Manual](https://github.com/riscv-non-isa/riscv-asm-manual/blob/master/riscv-asm.md), an expansion of `call` can look like this:

```
auipc   x1, 0x12345
jalr    x1, 32(x1)
```

So, as you said, `jalr` has rd=x1 but it also has rs1=x1. The current cfi_forward.adoc has this:

```
is_indirect_call_jump = ( (JALR || C.JR || C.JALR) &&
                          (rs1 != x1) && (rs1 != x5) ) ? 1 : 0;
is_sw_guarded_jump = ( JALR && rd == x7 && rs1 == x7 ) ? 1 : 0;
is_lp_expected = is_indirect_call_jump & ~is_sw_guarded_jump;
```

My reading is that `is_lp_expected` will be 0 because rs1=x1. Before the commit 157ebd a landing pad would be expected with `call`.

In contrast, the `tail` pseudoinstruction does currently require `lpad` at the target address since `tail` uses rs1=x6.

ved-rivos, 2023-06-16:
You are right. The qualification that `rd == x0` got missed. 
Should be:
    `is_ret = (((rs1 == x1) || (rs1 == x5)) && (rd == x0)) ? 1 : 0;`
    `is_indirect_call_jump = ( (JALR || C.JR || C.JALR) && ~is_ret ) ? 1 : 0;`

ved-rivos, 2023-06-16:
Addressed in PR  #122

Larhzu, 2023-06-17:
The final `)` is missing but otherwise the PR looks correct. The line is then fairly long, I didn't check how it looks as PDF.

Not a real suggestion, just thinking out aloud that it might look good this way:

```
is_lp_expected = ( (JALR || C.JR || C.JALR) &&
                   !(rd == x0 && (rs1 == x1 || rs1 == x5)) &&
                   !(rd == x7 && rs1 == x7) ) ? 1 : 0;
```

But keeping the helper variables is good, for example, if support for software guarded branches was optional. The [issue 84](https://github.com/riscv/riscv-cfi/issues/84) was about adding software guarded branches and it the linked to a [gcc thread](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=104816) and that linked to a [lkml post](https://lkml.org/lkml/2022/3/7/1068) which were good reads. On x86-64 the `notrack` prefix is similar to the software guarded branch but `notrack` support in the processor can be disabled so that a landing pad is always required, making CFI protection stronger. x86-64 Linux intentionally keeps `notrack` disabled in kernel mode. It is possible to create jump tables for `switch` statements with landing pads, and this can help against speculative execution problems better than software guarded branches (speculation may bypass jump table bounds checks).

Reading those links made me feel that the pros and cons of supporting software guarded branches should be understood well. I was reading the Zicfisslp spec for fun only and the value of software guarded branches is beoynd my skills and knowledge. So I won't comment it any further.

Thanks!

closed 2023-06-17 by ved-rivos