riscv/riscv-cfi #132: Excessive code space usage by LPAD instruction

brucehoult, 2023-07-16:
I've just noticed that a month ago `lpcll` and `lpcnl` were replaced by `lpad` and the encoding changed to use hint space in `auipc` instead of space in `system`.

https://github.com/riscv/riscv-cfi/issues/113

I don't object to using space from `auipc` but I do object to using ALL of it.

Reading the ISA manual I thought "wow, that's a lot of HINT space available". It's something close to 2.9 million code points. And most hint instructions are using a single code point each. The Non-Temporal Locality extension really splurges and uses 3 code points in the 32 bit space plus 3 code points in the C extension space.

The newly-proposed `lpad` instruction with 2^20 = 1,048,576 code points uses more than 36% of the entire HINT opcode space.

Is it really worth it?

The previous `lpcll` and `lpcnl` instructions used 1024 code points each. That seems entirely reasonable.

I understand that longer labels provide better security, and some people are willing to pay for better security.

Under the previous scheme, people who want to use 20 bit labels pay for it by using two instructions instead of one instruction at the landing pad. People who are happy with a 10 bit label or simply "yes, this is a landing pad" 0 bit label only use one instruction at the landing pad.

That seems reasonable to me.

High end implementations could fuse the `lpcll` and `lpcnl` instructions.

Under the new scheme, **everyone** pays the cost of 36% of the hint encoding space being gone forever, for a single instruction, whether they want CFI or not.

Using 0.07% of the hint encoding space for `lpcll` and `lpcnl` seems entirely reasonable.

Maybe an intermediate position would be tolerable, using 4.5% of the hint space by using half of one of the 2^17 code points hint spaces for each of `lpcll` and `lpcnl` with a 16 bit label. That is `addi` (minus 1 code point used for NOP), `andi`, `ori`, `xori`, `slti` or `sltiu` with rd=x0.

Using 36% of the hint space just doesn't seem justifiable to me. Has that been approved by the opcode space allocation people?

ved-rivos, 2023-07-16:
Use of 10-bit or even 16-bit labels does not provide enough label space for security. The typical label allocation scheme is to hash the prototype of the function. A typical program will need at least a 20-bit label space to meet goal of less than or equal to 10% hash collisions. Due to this for security the `lpcll` and `lpcnl` would be encoded together to provide a 20-bit label space. Considering this, the change to using AUIPC x0 was suggested by the ARC and approved by the opcode space allocation people [https://lists.riscv.org/g/tech-unprivileged/message/506]. The definition is that when Zicfilp is active, the `auipc` with `rd=x0` is a `lpad` and otherwise it reverts to being a no-op - for implementations that do not need CFI.

brucehoult, 2023-07-17:
> Use of 10-bit or even 16-bit labels does not provide enough label space for security.

That seems like a subjective value judgement.

The design clearly anticipates that some users will regard a simple wildcard "this is a landing pad" as being adequate, the same as e.g. `ENDBR64` on x86 or `HINT #34` on Aarch64, which as I understand it each use a single 32 bit code point.

A 10 bit label might not be perfect, but it seems like it's already 1024 times better than no label.

The proposal as it was prior to June 16 enabled those who want/need it to use 20 bit labels at the cost of one extra (fusable) instruction.

> the change to using AUIPC x0 was suggested by the ARC and approved by the opcode space allocation people

I'm surprised, tbh.

Yangff, 2023-08-20:
You can still create any hash size you want with even one code point by adding extra checkpoint afterward.. 
e.g., with CET: https://arxiv.org/ftp/arxiv/papers/2303/2303.16353.pdf

ved-rivos, 2023-08-21:
The Zicfilp extension avoids added code growth and to makes it low performance overhead even for simpler implementations. Doing explicit label checks in software leads to additional undesirable code growth and for simpler in-order implementations that code execution would lead to additional performance overheads. The landing pad by embedding the label minimizes the code growth, avoids a branch that needs to be predicted and minimizes the performance overhead even for simpler implementations that may not have out of order execution, macro-fusion.

With this `lpad` encoding there are still 1.5 million hint encodings reserved for standard use and 256K encodings available for custom use. The use of HINT  space is constrained - as you would note by the introduction of the Zimop/Zcmop. The architecture has currently a handful (5) hints. 

For ABI purposes we would want the label to be generated using a hash of the [mangled name](https://itanium-cxx-abi.github.io/cxx-abi/abi.html#mangling-builtin) . This would allow us to be compatible with the RISC-V psABI and be able to link binaries.  Generating random labels would break such ABI compatibility. To determine the number of unique prototypes in a program one  can  use the following command: find . -name "*.c*" | xargs -I{} clang -Xclang -ast-dump -fsyntax-only {} 2>&1 | grep FunctionDecl | grep -o "'.*'" | tr -d \' | tee -a fn_list.txt. This list can be put through `sort` and then `uniq`. The number of unique prototypes that can be found in larger bodies of  code - linux (181197), FFmpeg (14373), openssl (11269), freebsd(87664).

closed 2023-08-29 by ved-rivos