riscv/riscv-cfi #133: Misleading caption?: "`lpad` operation"

a4lg, 2023-07-18:
```
if (xFCFIE != 0)
    // If PC not 4-byte aligned then illegal-instruction
    if pc[1:0] != 0
        Cause illegal-instruction exception
    // If landing pad label not matched -> illegal-instruction
    else if (inst.LPL != x7[31:12] && inst.LPL != 0 && ELP == LP_EXPECTED)
        Cause illegal-instruction exception
    else
        ELP = NO_LP_EXPECTED
else
    no-op
endif
```

It explains not just the operation of `lpad` but a part of operations just after an indirect branch.
Since `lpad` is now a HINT, I suggest describing those operations as a part of an indirect branch and not "`lpad` operation" alone (it will make the `lpad` instruction *more like* HINTs).

ved-rivos, 2023-07-18:
The operations shown are all performed by the `lpad`. The `lpad` is encoded using `auipc` with `rd=0` but when fcfi is active it is no longer a  no-op - it is the `lpad` instruction. There is indeed an operation that determines if the fetched bytes are an `lpad` when `ELP` is `LP_EXPECTED` and faults when it is not - that is why it is not shown in the `lpad` operation since that happens when the code point is **not** `lpad`.

a4lg, 2023-07-24:
Okay.
I wondered how did you convince the ARC member to allow use of the HINT encodings (despite that there was a strong consensus **not** to use HINTs) and my interpretation was that "`lpad` is not a real instruction itself".
I should not have forced *my* interpretation to others.

closed 2023-07-24 by a4lg