riscv/riscv-cfi #136: Proposal: Interactions with compressed extensions

a4lg, 2023-08-04:
This is a clarification request and a proposal at the same time.

Looking at the specification, it seems it states that the compressed instructions are mandatory on Backward-edge control-flow integrity, considering the fact that `Zcmop` should depend on compressed instructions (either `C` or `Zca` extension (a minimum subset of `C`); may be simplified into the `Zca` extension only).

> The `Zicfiss` extension depends on the `Zicsr`, `A`, `Zimop`, and `Zcmop` extensions.

If compression instructions are mandatory to the `Zicfiss` extension, just close this issue.

If not, continue reading.

---

If the content of the `Zicfiss` extension should change depending on the existence of the compressed instructions, would you mind clarifying that?  I'll provide some technical advises.

1.  `Zicfiss` should not depend on `Zcmop` if compressed instructions are not supported.
2.  If you describe existence of the compressed instructions as *extensions*, prefer using words like "either `C` or `Zca` extension" (I'll detail this below).

Since the `Zca` extension is also a base of compressed instructions in [the code size reduction specification](https://github.com/riscv/riscv-code-size-reduction) (and a subset of `C`), this extension should be referenced if you describe "the compressed instructions" as extensions.

I'm currently working to make sure that the existence of the `C` extension implies `Zca` (because of its "subset" nature) so that just referencing the `Zca` extension is sufficient.  However, I'm currently facing some objections (claiming that `C` and `Zca` should be separate).

Meanwhile, referencing *both* `C` and `Zca` is safer.  Of course, if you just mention "the compressed instructions" not specifying the extension(s), that's okay.

ved-rivos, 2023-08-04:
Zcmop depends on C - so transitively Zicfiss depends on C. Thats why C was not in that list.

ved-rivos, 2023-08-04:
So yes, Zicfiss depends on C.

a4lg, 2023-08-04:
Understood.  So, there's no point opening this issue.  Thanks for reviewing!

closed 2023-08-04 by a4lg