riscv/riscv-cfi #143: Do coroutines need special handling?

jrtc27, 2023-08-17:
The RISC-V unprivileged spec has the following for JALR:

| rd is x1/x5 | rs1 is x1/x5 | rd=rs1 | RAS action |
|-|-|-|-|
| No | No | - | None |
| No | Yes | - | Pop |
| Yes | No | - | Push |
| Yes | Yes | No | Pop, then push |
| Yes | Yes | Yes | Push |

Extending this with rs1 is x7 and is_lp_expected gives, I believe:

| rd is x1/x5 | rs1 is x1/x5 | rd=rs1 | rs1 is x7 | RAS action | is_lp_expected |
|-|-|-|-|-|-|
| No | No | - | No | None | Yes |
| No | No | - | Yes | None | No |
| No | Yes | - | -  | Pop | No |
| Yes | No | - | No | Push | Yes |
| Yes | No | - | Yes | Push | No |
| Yes | Yes | No | - | Pop, then push | No |
| Yes | Yes | Yes | - | Push | No |

For forward edges you can therefore express:
* Protected tail call (rd != x1/x5, rs1 != x1/x5/x7)
* Unprotected tail call (rd != x1/x5, rs1 = x7)
* Protected call (rd = x1/x5, rs1 != x1/x5/x7)
* Unprotected call (rd = x1/x5, rs1 = x1/x5/x7, rs1 = x7/rd)
* Unprotected co-routine call (rd = x1/x5, rs1 = x1/x5, rs1 != rd)

What's missing is the ability to express a protected co-routine call. Is this intended? JALR goes out of its way to specify encodings for co-routines. In general co-routine resumption probably looks more shadow-stack-y than landing-pad-y, but co-routine starting isn't, and could be via a function pointer-like quantity.

ved-rivos, 2023-08-17:
The `rd = x1|x5, rs1 = x1|x5, rs1 != rd` hint is most useful case where two subroutines are continually "returning" to their resumption point. Doing a "pop followed by a push" on the return address prediction stack keeps the "resume" address of the coroutines resumption location at the top of the stack - without the pop it would underflow. So through these "return" ping pongs - which you rightly noted is "shadow stacky" they keep replacing the top of the return address prediction stack and when eventually they are done there is one misprediction but the rest of the return address prediction stack leading to this coroutine sequence is preserved. This was intended and nothing special was needed for this case.

jrtc27, 2023-08-17:
Yes. But what about the initial “call into a coroutine” case? That may need protection.

ved-rivos, 2023-08-17:
The initial call should be using a regular indirect call since the coroutine entry point would be address taken and this call will need to land on the landing pad. `swapcontext` on x86 for example differentiates between the initial case and the return case for that reason.

closed 2023-08-23 by ved-rivos