riscv/riscv-cfi #158: Implicit memory references to shadow stack pages should cause traps

kasanovic, 2023-10-03:
The document states that instruction fetches should cause an instruction page fault, but does not describe behavior of other implicit memory accesses. (e.g., page table walks, Svadu A/D bit updates).  I'm assuming these should also cause faults but should be specified.

ved-rivos, 2023-10-03:
Given that the shadow stack page is characterized as a virtual memory attribute, other implicit memory accesses fall outside the scope of this specification. It should be noted that the physical memory allocated for mapping a shadow stack page may also serve as a page table structure; however, such usage is not discernible within the context of this specification.

ved-rivos, 2023-10-07:
As presently defined, the `XWR=010` encoding continues to be reserved for future standard use in G-stage PTE. Is the intent to cause an access fault instead of a guest-page-fault for an `XWR=010` encoding in the G-stage PTE.

jhauser-us, 2023-10-14:
> Is the intent to cause an access fault instead of a guest-page-fault for an `XWR=010` encoding in the G-stage PTE.

The Architecture Review Committee discussed this recently and the answer is "no".  The ARC concluded that G-stage address translation should be unchanged and not cause any new access faults, even when a G-stage PTE's permission bits XWR are binary 010.  The new interpretation that Zicfiss gives to XWR = binary 010 should apply only to single-stage translation controlled by CSR `satp`, and to VS-stage translation controlled by CSR `vsatp`.

Because automatic reads and updates of PTEs in memory always involve either real physical addresses or guest physical addresses, those memory accesses will not cause access faults due to any PTE's XWR bits = binary 010.  Scratch that part of Krste's original description of this issue.

Two changes are requested for the Zicfiss document:

First, if not already done, it should be made clear that the new interpretation that Zicfiss gives to a PTE's bits XWR = binary 010 applies only to single-stage and VS-stage address translation and not to G-stage translation.  For G-stage translation, there will be no way to mark any pages as shadow stacks, and no distinction will be made between SS and non-SS memory accesses.  These distinctions occur only for single-stage and VS-stage translation.

Second, instead of having an instruction fetch from an SS page cause an instruction page fault, Zicfiss should specify that an implicit memory access that goes through full address translation (single-stage translation under the control of `satp` or VS-stage translation controlled by `vsatp`) causes an *access fault* if the implicit access is attempted to an SS page.  (As usual, if an access may be split into multiple pieces, such as a misaligned instruction fetch, then each piece may raise a different exception.)  We suggest also adding a nonnormative comment that, at the time this spec is written, the only implicit memory accesses that go through full address translation and thus can cause such SS-page access faults are instruction fetches.

ved-rivos, 2023-10-14:
> First, if not already done, it should be made clear that the new interpretation that Zicfiss gives to a PTE's bits XWR = binary 010 applies only to single-stage and VS-stage address translation and not to G-stage translation. 

This matches how its specified currently. The encoding is reserved in G-stage PTEs. There is also an informative note that a future extension may define the XWR=010b encoding.

> Second, instead of having an instruction fetch from an SS page cause an instruction page fault,

This change has been made to the specification. The updated text is as follows:
`Implicit accesses, including an instruction fetch, to the shadow stack page are not allowed. Such memory accesses cause an access-fault exception corresponding to the original access type.
`

ved-rivos, 2023-10-14:
> We suggest also adding a nonnormative comment that, at the time this spec is written, the only implicit memory accesses that go through full address translation and thus can cause such SS-page access faults are instruction fetches.

The following comment is included in PR #166 

> While the specification mandates that an access-fault exception shall be generated when either single-stage or VS-stage address translation is invoked for an implicit access targeting a shadow stack page, it is pertinent to highlight that, at the time of this specification's drafting, instruction fetches are the exclusive class of implicit accesses that are subjected to either single-stage or VS-stage address translation.

ved-rivos, 2023-10-17:
Suggested updates were merged into the document.

closed 2023-10-17 by ved-rivos