riscv/riscv-cfi #188: Ebreak at landing point

jklockars, 2023-12-01:
"The software-check exception caused by Zicfilp has higher priority than an illegal-instruction...", and at least the current version of the synchronous exception priority table in the priv-std claims that illegal instructions have the same priority as ebreak.

So, putting an ebreak at a landing point will cause a software-check exception instead of a breakpoint exception when Zicfilp is enabled. Is that the intention?
(It is still possible to set up an instruction address breakpoint for the same address, of course, if available.)

Obviously, software-check exception is not in the current table at all. This "problem" could be avoided if ebreak was moved to its own spot above the software-check exception (ie let ebreak be a "valid" landing pad).

ved-rivos, 2023-12-01:
That was the intention. The debuggers plan is to place the breakpoint at the the instruction following the landing pad. Please see [slide 8 and 9](https://lpc.events/event/17/contributions/1539/attachments/1189/2445/riscv_cfi_plumbers_2023.pdf) from the Linux plumbers conference.

jklockars, 2023-12-03:
OK.

closed 2023-12-03 by jklockars