riscv/riscv-cfi #190: Missing store/AMO access-fault in Listing 1~5 at Chapter 2 (Shadow Stack)

henry-hsieh, 2023-12-05:
In section 2.8, the spec said:

> If satp.MODE (or vsatp.MODE when V=1) is set to Bare and the effective privilege mode is below M, shadow stack memory accesses are prohibited, and shadow stack instructions will raise a store/AMO access-fault exception. When the effective privilege mode is M, any memory access by an SSAMOSWAP.W/D instruction will result in a store/AMO access-fault exception.

It's nice to have that the access-fault in operations (Listing 1~5) of shadow stack instruction operations.

For example, the Listing 1 could be modified to:
```
if (xSSE == 1)
    if satp.MODE == Bare && virtualization_mode == 0
        raise store/AMO access-fault exception
    else if vsatp.MODE == Bare && virtualization_mode == 1
        raise store/AMO access-fault exception
    endif
    mem[ssp - (XLEN/8)] = X(src) # Store src value to ssp - XLEN/8
    ssp = ssp - (XLEN/8)         # decrement ssp by XLEN/8
endif
```

ved-rivos, 2023-12-05:
The suggested modification to listing 1 is perhaps not right since access-fault are caused by the memory access made by the instruction - in this example the store to memory. The operations listing have followed the convention used by Sail and the convention for listing operations where the faults caused by memory access (*-access-fault, *-page-fault, *guest-page-fault) arising from page-based virtual memory system (including two stage address translation), PMP, and PMA checks are not shown as part of the operations performed by the instruction.

henry-hsieh, 2023-12-05:
I see. So that the operation is used by Sail model. I'll close the issue. Thanks!

closed 2023-12-05 by henry-hsieh