riscv/riscv-cfi #195: Rationale behind "use of Zicfiss in U-mode requires S-mode to be implemented. Use of Zicfiss in M-mode is not supported."

Silabs-ArjanB, 2023-12-06:
The RISC-V Shadow Stacks and Landing Pads" specifications states the following:

> Additionally, use of Zicfiss in U-mode requires S-mode to be implemented. Use of Zicfiss in M-mode is not supported.

Could you please explain the rationale behind not supporting Zicfiss in U-mode when S-mode is not present?

Similarly, what is the rationale for not supporting Zicfiss in M-mode?

Best regards,
Arjan

ved-rivos, 2023-12-06:
The Zicfiss provides a shadow stack that resists arbitrary overwrites and provides the `sspush`/`sspopchk` as the only sanctioned means to manipulate the shadow stack, thereby reducing the attack surface. When page-based virtual memory system is active, the extension provides a PTE encoding to designated virtual memory pages as shadow stack pages and enforce the protections. The extension is also not usable if the page-based virtual memory system is disabled by placing it in Bare more.

Use of Zicfiss M-mode and in M+U implementations, extensions are needed to the PMP and ePMP to designated shadow stack memory ranges. Further, the SPMP extension that is under development that enables the use of PMP like physical memory range based protections by S-mode. 

Support for extending Zicfiss for use in M-mode and for M+U implementations is expected to be developed subsequently as a follow on extension.

Silabs-ArjanB, 2023-12-06:
Hello @ved-rivos Thank you for your fast and detailed answer.

closed 2023-12-06 by Silabs-ArjanB