riscv/riscv-cfi #201: Does Zicfiss require Zcmop and thus Zca to be implemented?

topperc, 2023-12-28:
The spec includes this sentence "The Zicfiss extension has dependencies on the following extensions: Zicsr, Zimop, and Zcmop.".
One reading of this suggests that Zicfiss requires Zcmop to be implemented. And since Zcmop requires Zca, then Zca must also be implemented.

I doubt that this was the intent and expect that Zicfiss can be supported by hardware that does not support compressed instructions. And that use of compressed Zicfiss instruction require Zcmop, but uncompressed Zicfiss instructions do not.

Which interpretation is correct?

ved-rivos, 2023-12-28:
Zicfiss provides provides compressed form of `sspush x1` and `sspopchk x5`. The compressed instructions depend on Zcmop. However, if compressed instructions are not implemented then the compressed instructions provided by Zicfiss also do not need to be implemented. Further mandates may be placed by profiles but Zicfiss besides depending on Zcmop does not require implementing compressed instructions.

ved-rivos, 2024-01-02:
Hope that addressed the question.

closed 2024-01-02 by ved-rivos

topperc, 2024-01-02:
It did. Will it be clarified in the spec? I filed this because LLVM received a patch that interpreted the sentence as meaning -march=rv64i_zcfiss was equivalent to -march=rv64i_zcfiss_zca_zcmop.

ved-rivos, 2024-01-03:
Please see if this PR #202 clarifies this.

topperc, 2024-01-03:
> @topperc please see if PR #202 clarifies this.

Looks good to me.