riscv/riscv-cfi #203: Reset dcsr.PELP to NO_LP_EXPECTED upon resuming from debug mode?

mylai-mtk, 2024-01-05:
Regarding Zicfilp, the "Preserving Expected Landing Pad State on Traps" section:

> An MRET or SRET instruction is used to return from a trap in M-mode or S-mode, respectively. When executing an xRET instruction, if xPP holds the value y, then ELP is set to the value of xPELP if yLPE is 1; otherwise, it is set to NO_LP_EXPECTED; xPELP is set to NO_LP_EXPECTED.
> 
> Upon entry into Debug Mode, the pelp bit in dcsr is updated with the ELP at the privilege level the hart was previously in, and the ELP is set to NO_LP_EXPECTED. When a hart resumes from Debug Mode, if dcsr.prv holds the value y, then ELP is set to the value of pelp if yLPE is 1; otherwise, it is set to NO_LP_EXPECTED.

As one can see, `xPELP` is reset to `NO_LP_EXPECTED` upon `<M/S>RET`, but this behavior is not mandated on `dcsr.PELP` when resuming from debug mode. Is this discrepancy intended?

Thanks,
Ming-Yi Lai

ved-rivos, 2024-01-05:
Yes, that was intentional and follows similar difference between `dret` and `mret/sret` for `dcsr.prv` and `mstatus.mpp/spp`. Unlike M/S mode there is no scheduler in debug mode or a privilege level higher than Debug mode.

mylai-mtk, 2024-01-08:
Ok, thanks for the reply :)

closed 2024-01-08 by mylai-mtk