riscv/riscv-cfi #219: code optimization

ubc-guy, 2024-03-28:
The example given in the backward spec is:

    function_entry:
        c.addi sp,sp,-8  # push link register x1
        c.sd x1,(sp)     # on regular stack
        c.sspush x1      # push link register x1 on shadow stack
         :
        c.ld x5,(sp)     # pop link register x5 from regular stack
        c.addi sp,sp,8
        c.sspopchk x5    # fault if x5 not equal to shadow return address
        c.jr x5

Why not define the new instructions to absorb the addi instructions, eliminating bloat?

    function_entry:
        c.sspush x1      # push link register x1 on shadow stack AND if no fault addi sp,sp,-XLEN/8
        c.sd x1,(sp)     # push link register x1on regular stack
         :
        c.ld x5,(sp)     # pop link register x5 from regular stack
        c.sspopchk x5    # fault if x5 not equal to shadow return address AND if no fault addi sp,sp,XLEN/8
        c.jr x5

This would be faster and use less code. 

This would require the "maybe ops" version to have different behaviour (sp = sp+/-8 instead of sp=8), but the maybeops haven't been fully ratified yet (or maybe it's still stoppable/fixable).

sorear, 2024-03-28:
An earlier version of the spec had `sspop` which wrote to both ssp and to an integer register (with the loaded value). It was removed on the grounds that it was difficult for some microarchitectures to perform both in one operation.

ved-rivos, 2024-03-29:
This was not pursued due to:
1. Code size considerations
2. Execution time and implementation complexity considerations.
3. Consideration about encoding the use of `x2` as stack pointer in ISA

**Code size**
The +/- 8 is not always true for all functions. Some examples are as follows:
```
0000000000000f4e <Barrett_HR>:
     f4e:	716d                	addi	sp,sp,-272
     f50:	e606                	sd	ra,264(sp)
...
0000000000001888 <main>:
    1888:	b0010113          	addi	sp,sp,-1280
    188c:	4e113c23          	sd	ra,1272(sp)
...
00000000000007c8 <showval_128>:
     7c8:	7139                	addi	sp,sp,-64
     7ca:	fc06                	sd	ra,56(sp)
```
The size of the stack frame depends on local variables created on the stack by that function. This would require adding a 6-bit or a 12-bit immediate (compressed/non-compressed) source to the `sspush`. It is not possible to encode an immediate with Zcmop. Which would lead to `sspush` being always uncompressed and that removes the code size benefit of including the `sp` update into the instruction. For uncompressed form it would would need 8K code points to encode. 

**Execution time and complexity**
`sspush` is defined to need 2 sources and 1 destination. Including a `addi` functionality would require 3 sources, and immediate source, and 2 destinations. For most implementations this may lead to cracking or microcoding the instruction which makes the execution time benefits go away. Besides this adds adding significant complexity to the implementation.

**Embedding a convention**
Lastly, this design would require embedding the notion of `x2` being `sp` - which is a ABI/SW convention but is not required by the ISA into the ISA. Any register may be used as the stack pointer.

ved-rivos, 2024-04-26:
Closing this issue.  Adding an immediate to the instruction makes the use of compressed forms of `sspush` and `sspopchk` not possible. Using uncompressed form removes the code size benefits of absorbing the `c.addi` into these instructions. So the benefits of this are low besides adding implementation complexity and non-conformance with Zimop/Zcmop.

closed 2024-04-26 by ved-rivos