riscv/riscv-cfi #220: Did the ARC ever discuss changing the scope of menvcfg and henvcfg?

sorear, 2024-03-28:
henvcfg was originally added as part of the H extension to support giving _non-portable or non-privileged VS mode software_ an altered view of hardware. FIOM fits this scheme as it allows emulating hardware using access-fault or guest-page-fault traps. A guest which does not make hardware-specific instructions does not need henvcfg bits to run.

menvcfg was added at the same time as a measure for M-mode hypervisors and to allow the H extension to be emulated in M-mode. There is no reason to set menvcfg.FIOM unless the supervisor running in (H)S-mode is mismatched with the hardware. Importantly, a supervisor does not _know_ that it is mismatched with the hardware, so there is no reason to provide SBI access to menvcfg.FIOM; it is strictly an external property of the VM/SEE.

CBZE, CBCFE, CBIE follow the pattern closely, allowing nonportable or unprivileged code in a privilege mode to have its view of cache maintenance operations virtualized or restricted. A supervisor only performs CMOs which are correct for its own view of the world, so there is no reason to provide SBI access to menvcfg.{CBZE,CBCFE,CBIE}.

STCE, PMBTE, and ADUE also perform virtualization functions, allowing features to be disabled if they should not be exposed in a VM. All three are upward compatible at a supervisor level, so a properly written supervisor has no reason to interact with the bits and they have no reason to be exposed in the SBI. (As published, the RVA20S64 profile requires ADUE=0; this is clearly an erratum and nobody involved intended RVA20S64 to be read as licensing _software_ to assume A/D bit updates do not happen. My preferred resolution here is to publish the erratum and always set ADUE=1 in SEEs that support Svadu.)

SSE and LPE represent a **dramatic break** from prior practice in the henvcfg and menvcfg CSRs. They are the first bits in those CSRs that perform a function other than virtualization or access control, the first bits in those CSRs that ever make sense to have a non-fixed value for non-virtualized supervisors, and the first bits in those CSRs that require SBI access.

The situation is rather reminiscent of the VM field in mstatus in privileged spec 1.7 through 1.9.1, which affected supervisor execution in a way that supervisors needed to change and a SBI call was proposed as a mechanism for.

The ultimately adopted solution was to move VM into satp, becoming satp.MODE, and allowing supervisor software to control its own memory semantics.

Did the ARC ever discuss whether LPE and SSE were an appropriate use of the xenvcfg CSRs, or whether it would be more appropriate to move them (and [mh]envcfg.PMM) into a new register matching the semantics of rarely-written fields which control supervisor execution? If so, is there a link to it?

ved-rivos, 2024-03-29:
This was discussed by ARC and noted in minutes from [7/24/23](https://lists.riscv.org/g/tech-chairs/message/1539) 

Specifically,

> We don't want to use names "FCFIE" and "BCFIE" for enable bits because
> it's possible there could be future extensions for forward and backward
> CFI that are different from the current LP and SS.  We'd like to rename
> bit MFCFIE in mseccfg to MLPE, and rename bits *FCFIE and *BCFIE in the
> *envcfg CSRs to LPE and SSE respectively.  (Note that the other defined
> bits in the *envcfg registers don't have a leading 'S' or 'U' to
> indicate privilege mode.)
> 
> You inferred correctly from the ARC meeting minutes that we want to
> put most enable bits for LP and SS in the *envcfg registers instead of
> having the U-mode ones be in mstatus/sstatus.  We propose *envcfg bit 2
> be LPE and bit 3 be SSE.  (Bit 1 is expected to be an enable for the
> pointer-masking extension.)
> 
> 3.
> The name xFCFIE should be changed to xLPE, and the algorithm for
> determing xFCFIE can be replaced with a table for xLPE:
> 
>    Privilege mode   xLPE
> 
>       M             mseccfg.MLPE
>       S/HS          menvcfg.LPE
>       VS            henvcfg.LPE
>       U or VU       senvcfg.LPE
> 
> Likewise, the name xBCFIE should be changed to xSSE, and the algorithm
> for determing xBCFIE can be replaced with a table for xSSE:
> 
>    Privilege mode   xSSE
> 
>       M             1
>       S/HS          menvcfg.SSE
>       VS            henvcfg.SSE
>       U or VU       senvcfg.SSE

sorear, 2024-03-29:
That link is not accessible to ordinary RISC-V members, let alone the public; a public link for the 7/24/23 minutes is https://lists.riscv.org/g/tech-privileged/message/1428.

Moving the U-mode bits out of sstatus makes sense for the reasons discussed in the quote. Have the VS-mode and HS-mode bits been discussed in the context of SBI calls versus direct modification?

ved-rivos, 2024-03-29:
As the minutes note that the rubric was developed to define where to place configuration bits for lower privilege levels. The notes I listed in my previous post show the placement provided for the configuration bits in [m|h|s]envcfg - for each privilege level. The SBI calls and is required by the operating system -  in HS or VS mode - is enlightened and compiled to use these extensions. This enable is fairly static and done early in the startup based on a) whether the OS has been compiled for the feature b) the OS administrator has enabled the use of the feature. While such a control at the S/VS level could be a legitimate architectural option, the current approach of using the SBI offers similar functionality. And, in particular, given the rare need to modify this setting, the overhead introduced by an SBI call is insignificant and does not present a compelling reason to introduce additional  hardware arch functionality or introduction of new context switched state registers.

sorear, 2024-03-29:
Is it possible for me to request someone else to look at this and give an opinion?

ved-rivos, 2024-03-29:
Please do.

ved-rivos, 2024-04-26:
Posting link to further discussion on [Priv. list](https://lists.riscv.org/g/tech-privileged/message/1907).

ved-rivos, 2024-04-26:
Closing the issue.

closed 2024-04-26 by ved-rivos