riscv/riscv-cfi #223: SS instructions should report store/AMO guest  page fault due to VS stage memory  implicit access

leBsky, 2024-07-01:
Hi, the cfi_backward says:
[image]

However, the risc-v priviledged spec says:
[image]

My understanding is that the implicit memory access to support VS-stage translation checks load or store permission depending on if it writes or reads VS-stage page table. However, the final exception should always be store/AMO  guest page fault for SS instructions.

ved-rivos, 2024-07-03:
@leBsky - You are right that the exception should be reported as a store/AMO guest page fault. This note is incorrect and is removed in https://github.com/riscv/riscv-cfi/pull/222/commits/bd9938ca9cea2bfb72b64d90a6efc41e373a01a3

leBsky, 2024-07-03:
@ved-rivos Thanks, closed.

closed 2024-07-03 by leBsky