riscv/riscv-cfi #224: Clarifications about SS instructions in Machine mode

leBsky, 2024-07-04:
[image]
Here says " the Use of Zicfiss in M-mode is not supported? But what's the exact bahavior when execute Zicfiss instructions in Machine mode?

First, For sspush/sspopchk instructions, Should they be considered as Zipmop instructions?

Second, For ssamoawap.W/D instructions, here says:
[image]

Does it mean that as long as in machine mode, executing ssamoswap instrcution will cause store/AMO access fault?

ved-rivos, 2024-07-04:
The section 2.3 defines a term called xSSE that determines if Zicfiss is enabled at a privilege mode and the table 1 shows how xSSE is determined. Instructions that are encoded using MOPs/C.MOPs are listed in section 2.1. When xSSE is 0, these revert to their ZImop/Zcmop specified behavior. The behavior of ssamoswap is specified in section 2.7 by the pseudocode in listing 2.7. When MPRV is 1 all memory accesses in M-mode are with effective privilege determined by MPP. The MPRV behavior does not change for ssamoswap,

leBsky, 2024-07-04:
> The section 2.3 defines a term called xSSE that determines if Zicfiss is enabled at a privilege mode and the table 1 shows how xSSE is determined. Instructions that are encoded using MOPs/C.MOPs are [...]

Close, #225 solved this.

closed 2024-07-04 by leBsky