riscv/riscv-cfi #225: confused about shadow stack instructions when satp.mode == BARE and menvcfg.sse==0

leBsky, 2024-07-04:
it is said in spec that "Zicfiss instructions will revert to their behavior as defined by Zimop when menvcfg.sse == 0"

and it is also said in spec that "If satp.MODE (or vsatp.MODE when V=1) is set to Bare and the effective privilege mode is below M, shadow stack memory accesses are prohibited, and shadow stack instructions will raise a store/AMO access-fault exception."

Does it mean that  sspush/sspopchk will still generate store/AMO access fault  when in supervisor mode even though they revert their behavior as defined by Zimop because satp.mode is bare ?

kacouane, 2024-07-04:
Some Zicfiss instructions (ssamoswap) are not in the Zimop encoding space so they cannot revert to a Zimop-defined behavior.
 
This statement is here to expose the intended behavior of ssamoswap operations

leBsky, 2024-07-04:
> Some Zicfiss instructions (ssamoswap) are not in the Zimop encoding space so they cannot revert to a Zimop-defined behavior.
> 
> This statement is here to expose the intended behavior of ssamoswap operations

Let me elaborate it:

- when menvcfg.sse ==0 and satp.mode == bare,  sspush/sspopchk instruction revert to zimop-define bahavior, but ssamo instrcutions raise store/AMO access fault in supervisor mode.

- when menvcfg.sse ==1 and satp,mode == bare,  the sspush/sspopchk/ssamo instructions all generate store/AMO acces fault in supervisor mode.

Is is correct?

kacouane, 2024-07-04:
yes in my opinion this is the intended behavior

ved-rivos, 2024-07-04:
satp does not factor into determination of whether shadow stacks are enabled. The section 2.3 has table 1 that lists the conditions for determining xSSE. xSSE is always 0 for M-mode. When xSSE is 0, all instructions encoded using C.MOP/MOP as listed in section 2.1 revert to their Zimop/Zcmop defined behavior. 

ssamoswap is not encoded using a MOP. The behavior of ssamoswap and its pseudocode is listed in section 2.7. The ssamoswap can be executed in M-mode. 

The effective privilege of the memory accesses performed by ssamoswap - and any other load/store/amo - in M-mode is determined by MPRV and MPP. If MPRV is 1 then the effective privilege is determined by MPP. 

The section 2.8 states that if effective privilege is M then it will lead to a store/AMO access fault - as ssamoswap should only access shadow stack but there are no shadow stacks for M-mode. When effective privilege is not M, then whether virtual memory system is active at the effective privilege mode is determined by satp.MODE when V is 0 and by vsatp.MODE when V=1. As shadow stack instructions must access shadow stack and when virtual memory system is in Bare mode there are no shadow stacks, such an ssamoswap invoked in M-mode with effective privilege mode less than M will lead to an store/AMO access fault.

leBsky, 2024-07-04:
> oswap and its pseudocode is listed in section

That's very clear, Thanks. One more thing to confirm is that when menvcfg.sse ==1 and now is in supervisor mode, the sspush/sspopchk instruction should access virtual memory but if the satp.mode == bare, these instructions should also generate store/AMO access fault. Is it correct?

leBsky, 2024-07-04:
> satp does not factor into determination of whether shadow stacks are enabled. The section 2.3 has table 1 that lists the conditions for determining xSSE. xSSE is always 0 for M-mode. When xSSE is 0, all instructions encoded using C.MOP/MOP as listed in section 2.1 revert to their Zimop/Zcmop defined behavior.
> 
> ssamoswap is not encoded using a MOP. The behavior of ssamoswap and its pseudocode is listed in section 2.7. The ssamoswap can be executed in M-mode.
> 
> The **effective privilege mode** of the memory accesses performed by ssamoswap - and any other load/store/amo - in M-mode is determined by MPRV and MPP. If MPRV is 1 then the **effective privilege mode** is determined by MPP and MPV.
> 
> The section 2.8 states that if effective privilege is M then it will lead to a store/AMO access fault - as ssamoswap should only access shadow stack but there are no shadow stacks for M-mode. When effective privilege is not M, then whether virtual memory system is active at the effective privilege mode is determined by satp.MODE when effective V is 0 and by vsatp.MODE when effective V=1. As shadow stack instructions must access shadow stack and when virtual memory system is in Bare mode there are no shadow stacks, such an ssamoswap invoked in M-mode with effective privilege mode less than M will lead to an store/AMO access fault.

That's very clear, Thanks. One more thing to confirm is that when menvcfg.sse ==1 and now is in supervisor mode, the sspush/sspopchk instruction should access virtual memory but if the satp.mode == bare, these instructions should also generate store/AMO access fault. Is it correct?

ved-rivos, 2024-07-04:
When in S-mode the effective privilege mode of memory accesses is S and the same specifications apply, 

The case you stated will lead to a store/AMO access fault.

leBsky, 2024-07-10:
> satp does not factor into determination of whether shadow stacks are enabled. The section 2.3 has table 1 that lists the conditions for determining xSSE. xSSE is always 0 for M-mode. When xSSE is 0, [...]

One more question. Say it is in machine mode now and menvcfg.sse is 0, which means ssamoswap.w/d instructions are illegal in supervisor mode. However, now MPRV is 1, MPP is 1 and satp.mode is SV39 , so what's the behavior of executing ssamoswap.w/d in machine mode now? Will it raise illegal instruction exception? I think it should behave as normal memory access without illegal instruction.

leBsky, 2024-07-10:
it raise illegal instruction exception? I think it should behave as normal memory access without illegal instruction.

> satp does not factor into determination of whether shadow stacks are enabled. The section 2.3 has table 1 that lists the conditions for determining xSSE. xSSE is always 0 for M-mode. When xSSE is 0, [...]

One more question. Say it is in machine mode now and menvcfg.sse is 0, which means ssamoswap.w/d instructions are illegal in supervisor mode. However, now MPRV is 1, MPP is 1 and satp.mode is SV39 , so what's the behavior of executing ssamoswap.w/d in machine mode now? Will it raise illegal instruction exception? I think it should behave as normal memory access without raising illegal instruction.

More generally, should we consider menvcfg.sse/henvcfg.sse/senvcfg.sse when execute ssamoswap.w/d instruction in machine mode with MPRV is 1?

ved-rivos, 2024-07-10:
> what's the behavior of executing ssamoswap.w/d in machine mode now?

`SSAMOSWAP` will not raise an illegal-instruction exception and will attempt to make a shadow stack memory access with effective privilege mode of S in this example. Shadow stack instructions are restricted to accessing shadow stack (pte.xwr=010b) pages. Should a shadow stack instruction access a page that is not designated as a shadow stack page and is not marked as read-only (pte.xwr=001), a store/AMO access-fault exception will be invoked. Conversely, if the page being accessed by a shadow stack instruction is a read-only page, a store/AMO page-fault exception will be triggered.

leBsky, 2024-07-10:
> > what's the behavior of executing ssamoswap.w/d in machine mode now?
> 
> `SSAMOSWAP` will not raise an illegal-instruction exception and will attempt to make a shadow stack memory access with effective privilege mode of S in this example. Shadow stack instructions are restricted to accessing shadow stack (pte.xwr=010b) pages. To protect shadow stack memory, the memory is associated with a new page type – the Shadow Stack (SS) page – in the single-stage and VS-stage page tables. The encoding R=0, W=1, and X=0, is defined to represent an SS page. When menvcfg.SSE=0, this encoding remains reserved. Similarly, when V=1 and henvcfg.SSE=0, this encoding remains reserved at VS and VU levels. Should a shadow stack instruction access a page that is not designated as a shadow stack page and is not marked as read-only (pte.xwr=001), a store/AMO access-fault exception will be invoked. Conversely, if the page being accessed by a shadow stack instruction is a read-only page, a store/AMO page-fault exception will be triggered.

I see. Let me interpret it . Sine the menvcfg.sse is 0 now, it means the ptw.xwr 010 is still reserved in supervisor mode, so when in Machine mode MPRV is 1, MPP is 1, menvcfg.sse is 0 and satp.mode is not bare,  executing a ssamoswap.w/d instruction should  trigger a store/AMO page fault because of the reserved pte.xwr ==010  configuration.

ved-rivos, 2024-07-10:
Yes, if xwr=010 then it will lead to a samo page fault.

ved-rivos, 2024-07-11:
Hope that addressed the question.

closed 2024-07-11 by ved-rivos

leBsky, 2024-07-11:
> Hope that addressed the question.

Thanks, last call, Should reserved pte xwr==110 also triggers samo page fault under this circumstance?  
Let me list all the possibles and please help confirm if I understand correctly.

ssamoswap access memory, machine mode, mprv 1, mpp 1, mpv 0, menvcfg.sse 0
pte.xwr == 000/011/100/101/111, samo access fault
pte.xwr == 110/010/001, samo page fault

ssamoswap access memory , Machine mode, mprv 1, mpp 1, mpv 0, menvcfg.sse 1
pte.xwr == 000/011/100/101/110/111, samo access fault
pte.xwr == 001, samo page fault
pte.xwr == 010, pass

ved-rivos, 2024-07-11:
> ssamoswap access memory, machine mode, mprv 1, mpp 1, mpv 0, menvcfg.sse 0

pte.xwr == 011/100/101/111, samo access fault
pte.xwr == 000/110/001/010, samo page fault 

> ssamoswap access memory , Machine mode, mprv 1,  mpp 1, mpv 0, menvcfg.sse 1

Should be:
pte.xwr == 011/100/101/111, samo access fault
pte.xwr == 000/110/001, samo page fault
pte.xwr == 010, pass

leBsky, 2024-07-12:
> pte.xwr == 011/100/101/111, samo access fault pte.xwr == 000/110/001/010, samo page fault
>
>
> Should be: pte.xwr == 011/100/101/111, samo access fault pte.xwr == 000/110/001, samo page fault pte.xwr [...]

Very clear. Thank you very much!