riscv/riscv-cfi #232: Question about xPELP which store the elp state when trap happened

myx922, 2024-11-20:
Hi, I have some doubts about xPELP in status register. Zicfilp extension adds these fields to status register to store elp state when trap happend, and the extension also adds LPE field to xenvcfg to enable the extension. I have two questiones as follow:
1. For ssdbltrp extension, specs says when menvcfg.dte is zero, the implementation behaves as though Ssdbltrp is not implemented, and sstatus.SDT read only zero. However, for zicfilp, spec just says zicfilp adds LPE in menvcfg, and when LPE is set to 1, zicfilp extension is enabled, it's not mentioned that when the extension is disabled, the xPELP is read only 0. Why are these two extension implementations different?
2. When  Zicfilp extension is implemented but is disabled, if xPELP in status register should read only 0?

ved-rivos, 2024-11-20:
Unlike `SDT`, the `xPELP` does not affect behavior of the execution environment. Further, `LPE` is defined to disable the extension but the extension does  not behave as though its not implemented. Setting `LPE` to 0 in `menvcfg` does not cause `LPE` to be read-only zero in `henvcfg` or `senvcfg`. The state `xPELP` is part of the `xstatus` CSR which is already context switched by the OS/hypervisor leading to there being no need to protect OS/hypervisor against this bit being used as a covert channel. This contrasts with Zicfiss, where setting `menvcfg.SSE` will make `henvcfg.SSE` and `senvcfg.SSE` read only zero and prevent the `ssp` CSR from being accessed.

myx922, 2024-11-21:
> Unlike `SDT`, the `xPELP` does not affect behavior of the execution environment. Further, `LPE` is defined to disable the extension but the extension does not behave as though its not implemented. [...]

Thank you very much for your answer.

closed 2024-11-21 by myx922