riscv/riscv-cfi #233: If xLPE=0，should xPELP be cleared when executing xRET instruction?

Baixyzz, 2024-11-21:
In Zicfilp spec，we can see “An MRET or SRET instruction is used to return from a trap in M-mode or S-mode, respectively. When
executing an xRET instruction, xPELP is set to NO_LP_EXPECTED.”From this description, we can't see any description about xLPE.If mseccfg.MLPE =0 and mstatus.MPELP =1,should mstatus.MPELP keep its original value or  set to NO_LP_EXPECTED when executing mret instruction?

ved-rivos, 2024-11-21:
The behavior specified is as follows:
> An MRET or SRET instruction is used to return from a trap in M-mode or S-mode, respectively. When executing an xRET instruction, if the new privilege mode is `y`, then ELP is set to the value of `xPELP` if `yLPE` is 1; otherwise, it is set to `NO_LP_EXPECTED`; `xPELP` is set to `NO_LP_EXPECTED`.

The last statement states that `xPELP` is unconditionally set to `NO_LP_EXPECTED`

Baixyzz, 2024-11-22:
> The behavior specified is as follows:
>
>
> The last statement states that `xPELP` is unconditionally set to `NO_LP_EXPECTED`
@ved-rivos Thank you very much for your answer.But when I use the latest version of spike to simulate the scenario I mentioned, I see that mstatus.MPELP remains LP_EXPECTED after executing the mret instruction.So I want to know if SPIKE's handling is wrong？

ved-rivos, 2024-11-22:
The code in spike seems to set the `MPELP` unconditionally to `NO_LP_EXPECTED`. Do you know why the statement 4 is not taking effect?
```

1. if (ZICFILP_xLPE(prev_virt, prev_prv)) {
2.   STATE.elp = static_cast<elp_t>(get_field(s, MSTATUS_MPELP));
3. }
4. s = set_field(s, MSTATUS_MPELP, elp_t::NO_LP_EXPECTED);

```

Baixyzz, 2024-11-25:
Sorry, upon further inspection the spike version I was using was not the latest. I will close this issue.

closed 2024-11-25 by Baixyzz