riscv/riscv-cfi #238: hlvx and shadow stack pages

jiahzhang, 2024-12-10:
How does hlvx interact with shadow stack pages? The spec says
```
However, the shadow stack is readable by all instructions that only load from memory.
```
and
```
Instructions HLVX.HU and HLVX.WU are the same as HLV.HU and HLV.WU, except that execute permission takes the place of read permission during address translation. That is, the memory being read must be executable in both stages of address translation, but read permission is not required.
```

Does the hlvx requirement supersede the shadow stack text?

ved-rivos, 2024-12-10:
The HLVX instruction tests for execute permission in the page tables. Shadow stack pages are encoded with XWR=010 and attempts to access shadow stack pages using HLVX will lead to a load page fault.

jiahzhang, 2024-12-10:
Thanks for clarifying.

closed 2024-12-10 by jiahzhang