riscv/riscv-cfi #241: When virtualization is enabled, and the laod instruction accesses the shadow stack page table, does the g-stage require read-write permissions?

chara811, 2025-01-06:


ved-rivos, 2025-01-06:
Yes. Please see this text in the specification: 

> The G-stage address translation and protections remain unaffected by the Zicfiss extension. The xwr == 010b encoding in the G-stage PTE remains reserved. When G-stage page tables are active, the shadow stack instructions that access memory require the G-stage page table to have read-write permission for the accessed memory; else a store/AMO guest-page fault exception is raised.

chara811, 2025-01-06:
I still have some areas of confusion.
When the shadow stack instruction accesses memory, the g-stage requires read-write permissions. Does the g-stage also need read-write permissions when accessing the shadow stack page with a normal load instruction? This is not clearly stated in the spec.
Or, for the shadow stack page , is it mandatory for the g-stage to have read-write permissions, and if the g-stage only has read permissions, should it report guest page fault?When the shadow stack instruction accesses memory, the g-stage requires read-write permissions. Does the g-stage also need read-write permissions when accessing the shadow stack page with a normal load instruction? This is not clearly stated in the spec.
Or, for the shadow stack page , is it mandatory for the g-stage to have read-write permissions, and if the g-stage only has read permissions, should it report guest page fault?

ved-rivos, 2025-01-06:
> When the shadow stack instruction accesses memory, the g-stage requires read-write permissions. Does the g-stage also need read-write permissions when accessing the shadow stack page with a normal load instruction? This is not clearly stated in the spec.

The specification states: "the shadow stack instructions that access memory require the G-stage page table to have read-write permission". It does not state that normal load instruction needs read-write permission.

chara811, 2025-01-06:
So, can I understand that when virtualization is enabled, for the vs-stage page table, pte.xwr = 3b010, and for the g-stage page table, pte.xwr = 3b001, it is possible to configure these settings, but if an shadow stack instruction accesses it, it will cause a guest page fault, while a regular load instruction can access it normally, right?

ved-rivos, 2025-01-06:
yes

chara811, 2025-01-06:
Okey,thank you very much.

ved-rivos, 2025-01-06:
Feel free to ask if there are further questions.

closed 2025-01-06 by ved-rivos