riscv/riscv-cfi #244: When the ss instruction accesses a page table entry with xwr=3b001 and the memory attribute is non-idempotent, should it report  a store/AMO access-fault fault or a store/AMO page-fault ？

chara811, 2025-05-14:


kacouane, 2025-05-14:
Hello, for the priority of errors, you should refer to the privilege spec's "virtual address translation process" section.

So the fault given is the one according to shadow stack access protection (Page fault)

Anyway, you could understand this way: 
PMA rules must apply to the Physical address, so the translation must have been completed successfully before (pte data are virtual memory attributes)

[image]

chara811, 2025-05-14:
Thank you, that's very clear.

chara811, 2025-05-15:
I think that when there is a store/AMO access-fault, the page table is an erroneous page table. When encountering PTE.A = 0 or SSPUSH instruction with PTE.D = 0, hardware updates for AD should not be performed. Therefore, the priority of SS access-fault should be higher than the priority of AD updates. Is my understanding correct?

chara811, 2025-05-15:
Ok, thank you very much.